BruceDevices/firmware #47: UK keyboard

brettputer, 2024-06-15:
Anyone managed to find a UK Keyboard layout,  \ translates to # seems to be the main issue when using EN_US

bmorcelli, 2024-06-16:
Ok, I'll look into it..

mtclpt, 2024-06-30:
I can suggest this project to help develop the functionalities of:
-Wireless BadUSB over a WiFi AP
-BadUSB on M5StickC (..WiFi Duck and utilizes the native USB function of ESP32 S2/3 chip. As a result, you can run WiFi Duck with a default module below 10$..)
-Add Keyboard layouts

https://github.com/wasdwasd0105/SuperWiFiDuck

bmorcelli, 2024-07-27:
[Bruce3_Cardputer.zip](https://github.com/user-attachments/files/16401613/Bruce3_Cardputer.zip)
@brettputer Can you please validate the Bad Usb with the "en-UK" keyboard, please?

bmorcelli, 2024-07-28:
implemented[ here ](https://github.com/pr3y/Bruce/pull/95)

closed 2024-07-28 by bmorcelli