BruceDevices/firmware #52: probleme evil portal

ZOZO25, 2024-06-21:
bonjour j'ai tester l'evil portal il est super mais a des moment randome il me mes des caractère blizzard sur mon navigateur tel que :
P1: 1719561562 
P2: 404 
P3: 2 
P4: hsopY03ZDj4DjAySJWbA0Rw1igPU+jWL1npy87eM382DjQnchwd5j5DV9qclTGUja3pFApk6H6PN/MVhdf8NPw==

capture d'écran a l'appui 
[image]
[image]

ZOZO25, 2024-06-21:
[image]

bmorcelli, 2024-06-21:
Bruce, afin d'utiliser tout type de formulaire HTML, enregistre toutes les informations envoyées comme argument pour les requêtes POST et GET... probablement votre navigateur essaie d'envoyer ces informations quelque part et Bruce les capture... vous pouvez Pouvez-vous confirmer si vous utilisez EvilPortal de Bruce ou un code HTML personnalisé ?  S'il est personnalisé, pourriez-vous m'envoyer le fichier pour que je le voie ?

ZOZO25, 2024-06-21:
Bonjour c'est le code Bruce que j'utilise je l'ai mis sur mon M5 stick c plus 2 via burn le logiciel de M5 stack

Et je tien à presser que ça me fait ça que de temps en temps

DittmannCz, 2024-06-25:
Hello, is it possible to share information: name of the creds file and structure of the html? I tried many examples, chatgpt... nothing works, every time i log in - i still se number 0 and no record on SD

ZOZO25, 2024-06-25:
Bonjour, c'est pour un nouveau problème?

DittmannCz, 2024-06-26:
Well - I am not sure,shall I do anything?

DittmannCz, 2024-06-26:
I really miss any tutorial how to properly setup evil portal

bmorcelli, 2024-07-27:
will be fixed here https://github.com/pr3y/Bruce/pull/94

closed 2024-07-27 by bmorcelli