BruceDevices/firmware #54: Can you add the universal remotes that are used in the flipper zero. Plz read this. (pr3y)

Z-SHOCK-E, 2024-06-21:
https://github.com/flipperdevices/flipperzero-firmware/tree/dev/applications/main/infrared/resources/infrared/assets

Here are the files that are used by the flipper zero. There are multiple power-off commands in tv.ir file. I think that there is a script that sends all these power-off commands in the tv.ir file when the universal TV remote is used and when the power-off function is used in the universal AC remote it sends all the power-off commands in the projector.ir file and vice versa. So if you can create four scripts for the files (ac.ir ,  audio.ir ,  TV.ir ,  projector.ir) that sends all the power off commands when the power off function is selected and all the "volume +" commands when the "volume +" function is selected.  

I think there are multiple power-off functions in one file because different devices use different protocols and different protocols can't be in one command. ( In a .ir file )

I hope this will be another cool feature in the next update.

vla-droid, 2024-06-24:
IRDB would be better, but they are working on it.

Z-SHOCK-E, 2024-06-27:
But these files have all the required commands for ACs, Projectors, Audio systems and TVs plus it'd be much easier to fit inside the LittleFs on a m5stickc plus 2. Your idea of using IRDB is a great idea considering how many devices it'll control but I think it's more suited for the cardputer than the m5stick varients plus it'll take more effort to write the script that'll send all the IR code for a specific function. 

My idea is to add these files to the memory of the m5stickc plus 2 and four universal remote application for ACs, Projectors, Audio systems and TVs. When the universal remote for projectors is selected there will be selection of common projector commands (that're included in the Projector.ir file) . 
Ex: when the power off command is selected in the "Universal remote for projectors" application it'll send all the power off command included in the projector.ir file. (There multiple power off commands)

rennancockles, 2024-08-01:
You can do it yourself if you like. Bruce already has an option to load if commands from file. If you create a file with the commands that you want you can open it and send one by one or send all at once. 

You could take the ac.ir for example and delete everything but the OFF codes. Once you open that file on Bruce you can send all OFF commands.

We also have some example files in the `sd_files` folder.

closed 2024-08-23 by bmorcelli