BruceDevices/firmware #55: Question BadUSB

softm5, 2024-06-22:
Hi,
there is no badUSB with the Bruce software on the m5 stick c plus 2 why?
Thanks :)

bmorcelli, 2024-06-22:
StickCs doesn't have OTG to run as HID device..

It will need to plug another device to make it work.. I bought it to develop, but didn't make any progress yet

softm5, 2024-06-23:
with "marauder" I use this image which allows me to connect my sd card in gpio.
The problem is with “Bruce” it doesn’t work unfortunately.
probably because it was not programmed with it.

[image]

bmorcelli, 2024-06-23:
Bruce uses the same pins Marauder, Launcher and Nemo.. which is the same as the picture you sent.. 

Is your SDCard being read in the Launcher?

Remember to use an SDCard formatted in FAT32 and lesser than 32gb.. otherwise most of the firmwares doesn't recognize

softm5, 2024-06-23:
it's good the SD card works on the firmware.
What is OTG run HID device ?
(I don't understand.)
is it possible to do badusb via bluetooth directly?
THANK YOU

softm5, 2024-06-25:
Ce qui serait bien de faire sur "Bruce" pour le (BadUSB), est avec le bluetooth est de créer un hospot via m5 stick c plus 2, ainsi lorsque la connexion s'établie entre les deux, qu'on puisse se connecter sur le m5 stick via navigateur web et faire des commandes ou importer directement un payload.

bmorcelli, 2024-06-25:
This is indeed a great Idea... maybe use the WebUI to deploy these kind of "attacks"

mtclpt, 2024-06-30:
I can suggest this project to help develop the functionalities of:
  -Wireless BadUSB over a WiFi AP
  -BadUSB on M5StickC (..WiFi Duck and utilizes the native USB function of ESP32 S2/3 chip. As a result, you can run WiFi Duck with a default module below 10$..)
  -Add Keyboard layouts

https://github.com/wasdwasd0105/SuperWiFiDuck

softm5, 2024-06-30:
Pourtant avec le m5 stick c plus 2 on est quand même obligé d'acheter ce module à 10$ ?
Merci

mtclpt, 2024-06-30:
Hi softm5, No.
What I meant to say is that in this project they managed to use the regular USB port of several ESP32 modules to make BadUSB.

softm5, 2024-06-30:
Ok, ça peut être une idée aussi en passant par un AP WiFi. 👍

bmorcelli, 2024-07-02:
> -BadUSB on M5StickC (..WiFi Duck and utilizes the native USB function of ESP32 S2/3 chip. As a result, you can run WiFi Duck with a default module below 10$..)

WifiDuck uses ESP32-S2 or S3 ( same as Cardputer)

StickCs GPIO is connected to a CH9102 chip, that converts from UART to TTL (Serial), so it's not possible (yet, at least?) to use it as HID (Human Interface Device)..

ESP32-S2 and S3 have a OTG (On-The-Go) chip, that allows us to use it as HID (Mouse, Keyboard or whatever)

The solution to use BadUSB on StickCs would be connecting it to a UART->HID adapter, like [this](https://a.aliexpress.com/_msiWB9Q) for example... I bought one of these to test, but didn't have time to study and test

The other option is BLE keyboard, that's nice too, and easy to add to the project, I think...

PhucHauDeveloper, 2024-08-19:
> WifiDuck uses ESP32-S2 or S3 ( same as Cardputer)
>
> StickCs GPIO is connected to a CH9102 chip, that converts from UART to TTL (Serial), so it's not possible (yet, at least?) to use it as HID (Human [...]

i found [this](https://lang-ship.com/blog/work/m5stickc-attiny85-usb-keyboard/), attiny85 is pretty cheap and can make it into hat for stickc2 easily.

closed 2024-10-05 by bmorcelli