BruceDevices/firmware #57: RFID function doesn't work

kaisummer, 2024-06-24:
says "write failed" when i try to write to a blank card.

softm5, 2024-06-28:
J'ai un module RFID U031-B pour le m5 stick c plus 2 et cela ne fonctionne pas pour écrire sur une carte.
J'ai l'erreur "Writing failed"

rugosi, 2024-06-30:
> J'ai un module RFID U031-B pour le m5 stick c plus 2 et cela ne fonctionne pas pour écrire sur une carte. J'ai l'erreur "Writing failed" Est il possible de le rendre compatible ? Merci

J'ai la même expérience. C'est peut-être la sélection de la carte qui manque.

bmorcelli, 2024-07-29:
Can you plese confirm if this problem still exist using the Tag-o-matic function?

rennancockles, 2024-07-29:
Are you sure your tag is a writable one? Not all tags are writable.

softm5, 2024-07-29:
J'utilise des cartes qui peuvent lire/écrire
J'ai une erreur "error writting UID to tag" avec TAG-O-MATIC.

rennancockles, 2024-07-29:
Quel type de carte et quelles valeurs sak sont affichés lorsque vous utilisez TAG-O-MATIC pour lire votre carte?

softm5, 2024-07-29:
"CARTE MIFARE 1KB"
(SAK 08)

kaisummer, 2024-07-29:
> Are you sure your tag is a writable one? Not all tags are writable.

I'm pretty sure it is, it came with a pn532 module (I am not using the module, only the card that came with it)

IncursioHack, 2024-08-12:
The difference between blank RFID tags and RFID tags with a rewritable block 0 is essential to understand how these devices work and what applications they are best suited for. Let me explain each one:

1. Blank RFID Tags
Description: These are RFID tags that come from the factory without any information stored in the memory blocks or with generic information. The term "blank" generally refers to the fact that these tags are "empty" and ready to be programmed or written to for the first time.

Block 0: In many RFID systems, block 0 is where the UID (Unique Identifier) of the tag is stored. In standard RFID tags, the UID is programmed at the factory and cannot be changed.

Common Use: These tags are used in applications where security is not a major concern, and where the fixed, unchangeable UID serves as a unique identity for the tag.

2. RFID Tags with Rewritable Block 0

Description: These are special RFID tags where block 0 (where the UID is usually stored) can be rewritten or changed. This means that the UID is not fixed and can be cloned or modified by someone with access to the tag and the necessary equipment.

Security Risk: Tags with rewritable block 0 can be used to clone other RFID tags. This can be a problem in systems where security is based on the tag's unique UID, such as in access control systems.

Common Use: They are commonly used by enthusiasts and security professionals for testing, cloning, and security experiments. They can also be used in systems where the UID needs to be updated or customized after manufacturing.

rennancockles, 2024-08-13:
@kaisummer I just got my PN532 module and both the card and the tag that comes with it are not clonable. They do not support writing into the block 0, which is required to clone an rfid tag.

rennancockles, 2024-08-13:
@kaisummer you need something like this https://pt.aliexpress.com/item/1005006787338686.html

closed 2024-08-23 by bmorcelli