BruceDevices/firmware #59: Hi some feedback on the project

cinouw, 2024-06-27:
Hi so for the past week I've been using bruce 1.0.2 and here is the things that I've noticed 

- rf module support and spectrum feature
I tried using it first with stx882 TX module and srt882s RX module and it didn't work so I switched to the module that you've mentioned in the supported module list (FS1000A RF433T) module and it didn't work too so that is the first problem 

Note : based on my research any single pin protocol rf433mhz module is compatible with the cardputer because at it's core they all work the same by outputting or receiving digital/analog signal 

-RFID module support 
In this case also I used one of the modules that you've listed I used Mc rf522 and even after making sure that all the connections are done properly it didn't recognize it and I couldn't use the raid features so that needs fixing too

The last part is about the software 

-ui color change is not stored and after restart it resets

-wardriving could be ported from marauder firmware it wouldn't bee a nice feature 

-better ir receiving and ir transmitting ui could be made or ported from known firmware

-VPN client mode could be added by porting avaliable firmwares

Note : for better troubleshooting and experience note all the connections that are needed to be made for the modules in the wiki page so we can make sure that we did the same thing that you've done to get the module working

Thanks for the work it's really the greatest project on this thing the cardputer and we want to see it getting better and better 🫡🥰

mtclpt, 2024-06-30:
Hi, I agree that the connection diagram for the various modules would be very useful, I also couldn't get the CC1101 to work.

cinouw, 2024-07-01:
> Hi, I agree that the connection diagram for the various modules would be very useful, I also couldn't get the CC1101 to work.

I also couldn't get cc1101 to work but I think it's because of the limited pins available
Cc1101 connect over spi protocol and the only spi bus on the cardputer is connected to the sd card port i tried soldering wires to the pins and try to use it but after everything that I've done i couldn't get it to work and i think that's because of the limited i/o

closed 2024-07-29 by bmorcelli