BruceDevices/firmware #62: Connecting rfid and other stuff

Kcual, 2024-07-06:
Hello everyone, please tell me how to connect RFID-RC522 to M5Stick CPlus1.1. Also I would like to know about connecting FS1000A and card reader, is it possible to make it all work together (Rfid, FS1000A transmitter and card reader)? Does anyone have connection diagrams?

Niximkk, 2024-07-27:
It is possible to use all of them together, but it's not that simple. Since the RFID module uses the Grove port, you won't need any modifications. Now, for the SD card and FS1000A transmitter, you can try using a protoboard, some female pins, and software changes, since you will need to use one at a time.
--- 

The FS1000A transmitter is plug and play, GND 5V G26.

[image]

For the sd card, follow this:

| M5StickC | SD   |
|----------|------|
| 3V3      | 3V3  |
| GND      | GND  |
| 0        | CLK  |
| 36       | MISO |
| 26       | MOSI |
| GND      | CS   |

Now for the RFID module, the only one supported afaik is the [Official M5Stack one](https://shop.m5stack.com/products/rfid-unit-2-ws1850s). Which uses the grove port (i don't think you need a tutorial for that). 

## I hope this helps :3

Kcual, 2024-07-27:
Thanks a lot

Kcual, 2024-07-27:
Thanks a lot

closed 2024-07-27 by Kcual