BruceDevices/firmware #63: Building the project does not work

Petitoursbr1, 2024-07-09:
Since I saw the project, I tried to modify it so that it suits me better, but building the project does not work for me, even without making any changes to the code. Here is the output I get (btw sorry for my bad English I am french) :

At first I click there :

[image]

and I get this output :
```
*  Executing task: C:\Users\Ours\.platformio\penv\Scripts\platformio.exe run --environment m5stack-cardputer 

Processing m5stack-cardputer (platform: espressif32; board: m5stack-stamps3; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stack-stamps3.html
PLATFORM: Espressif 32 (6.7.0) > M5Stack StampS3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WireGuard-ESP32 @ 0.1.5
|-- IRremoteESP8266 @ 2.8.6
|-- Time @ 1.6.1
|-- LibSSH-ESP32 @ 4.2.0
|-- PCA9554 @ 0.0.1
|-- ESPping @ 1.0.4+sha.2bf42df
|-- NTPClient @ 3.2.1
|-- Timezone @ 1.2.4
|-- ESP32Time @ 2.0.6
|-- LiteLED @ 1.2.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- EEPROM @ 2.0.0
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
|-- Bad_Usb_Lib
|-- utility
|-- TFT_eSPI @ 2.5.43
|-- ESP32 BLE Arduino @ 2.0.0
|-- DNSServer @ 2.0.0
|-- WebServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Building in release mode
Compiling .pio\build\m5stack-cardputer\src\TV-B-Gone.cpp.o
Compiling .pio\build\m5stack-cardputer\src\bad_usb.cpp.o
Compiling .pio\build\m5stack-cardputer\src\ble_spam.cpp.o
Compiling .pio\build\m5stack-cardputer\src\clients.cpp.o
Compiling .pio\build\m5stack-cardputer\src\display.cpp.o
Compiling .pio\build\m5stack-cardputer\src\dpwo.cpp.o
Compiling .pio\build\m5stack-cardputer\src\evil_portal.cpp.o
Compiling .pio\build\m5stack-cardputer\src\globals.cpp.o
Compiling .pio\build\m5stack-cardputer\src\led.cpp.o
Compiling .pio\build\m5stack-cardputer\src\main.cpp.o
Compiling .pio\build\m5stack-cardputer\src\mfrc522_i2c.cpp.o
Compiling .pio\build\m5stack-cardputer\src\mykeyboard.cpp.o
Compiling .pio\build\m5stack-cardputer\src\openhaystack.cpp.o
Compiling .pio\build\m5stack-cardputer\src\rf.cpp.o
Compiling .pio\build\m5stack-cardputer\src\rfid.cpp.o
Compiling .pio\build\m5stack-cardputer\src\scan_hosts.cpp.o
Compiling .pio\build\m5stack-cardputer\src\sd_functions.cpp.o
Compiling .pio\build\m5stack-cardputer\src\settings.cpp.o
Compiling .pio\build\m5stack-cardputer\src\sniffer.cpp.o
[... 435 lines cut ...]
Compiling .pio\build\m5stack-cardputer\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\m5stack-cardputer\FrameworkArduino\wiring_shift.c.o
C:/Users/Ours/.platformio/packages/framework-arduinoespressif32@3.20014.231204/cores/esp32/esp32-hal-uart.c: In function 'uartSetPins':
C:/Users/Ours/.platformio/packages/framework-arduinoespressif32@3.20014.231204/cores/esp32/esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
C:/Users/Ours/.platformio/packages/framework-arduinoespressif32@3.20014.231204/cores/esp32/esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
      ^~~~~~~~~~~
Archiving .pio\build\m5stack-cardputer\libFrameworkArduino.a
Linking .pio\build\m5stack-cardputer\firmware.elf
c:/users/ours/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/Ours/.platformio/packages/framework-arduinoespressif32@3.20014.231204/tools/sdk/esp32s3/lib\libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; .pio/build/m5stack-cardputer/src/wifi_atks.cpp.o:C:\Users\Ours\Desktop\Bruce-main/src/wifi_atks.cpp:22: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\m5stack-cardputer\firmware.elf] Error 1
=============================================================================== [FAILED] Took 147.27 seconds ===============================================================================

Environment        Status    Duration
-----------------  --------  ------------
m5stack-cardputer  FAILED    00:02:27.266
=========================================================================== 1 failed, 0 succeeded in 00:02:27.266 ===========================================================================
 *  The terminal process "C:\Users\Ours\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'm5stack-cardputer'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.
```
Could someone help me ?

Petitoursbr1, 2024-07-09:
Just saw the wiki does work now, sorry 👍

bmorcelli, 2024-07-09:
Nice!! It was missing the Death bypass..

I'm glad you fixed it! Call us on Discord if you need

closed 2024-07-09 by bmorcelli