BruceDevices/firmware #66: Add CustomRF / SubGhz

eadmaster, 2024-07-16:
compatibile with flipper sub files
https://github.com/Zero-Sploit/FlipperZero-Subghz-DB

possibly reuse [code from my repo](https://github.com/eadmaster/Bruce/commit/b1f1c424e1118d1df9a09d5c9ec9e1d5db574409).

bmorcelli, 2024-07-17:
what's up @eadmaster ,I really liked your suggestions and I've spoken to pr3y about what you are doing.

We are expecting your Pull Request to merger your Serial CLI solution into Bruce in the next update.

Feel free to contact us on Discord! @bmorcelli or Pirata#5263

eadmaster, 2024-07-17:
no problem, i still have to test if the changes i've made did break some other feature (e.g. badUSB)

Regarding this issue, maybe i've found [something here](https://github.com/sui77/rc-switch/pull/170), i will make try to see if it works.

bmorcelli, 2024-07-17:
Nicee!!! I Pulled your repo this morning and I'm making changes over it regarding other Settings and other stuff..

eadmaster, 2024-07-19:
I've managed to add an **experimental** implementation via a new "Custom Subghz" menu, but has some important limitations: 

 - only 433mhz freq. is supported, due to the [RF433T module](https://docs.m5stack.com/en/unit/rf433_t)
 - only ASK/OOK supported, for the same reason

Switching to a CC1101  would remove these  limitations, but there is no easy way to connect it via a Groove connector afaik.

In order to test if my current implementation is working correctly i need either:

 - `rtl_433 -a 4` output of one of the remote signals in [this repo](https://github.com/Zero-Sploit/FlipperZero-Subghz-DB)
 - a raw `.sub`  file ripped with a FlipperZero from a remote i own (look for the lines containing `RfSend` in [this file](https://github.com/eadmaster/dotfiles/blob/main/bin/power))
- a donation to buy a FlipperZero myself (prolly too much to ask, but will try anyway)

As a bonus, i've added a "Recent" menu to quickly select previously sent commands, but it is still not working correctly (suggestions are welcomed).

Petitoursbr1, 2024-07-19:
Hi Mr. Eadmaster,

On this project [](https://github.com/bmorcelli/io433), you can find a diagram for the connection of a CC1101 to an M5Stick C+.

[image]

I don't know if it helps, but I'm trying because I'm very hyped about the project.

eadmaster, 2024-07-19:
There are no free Gpio pins on the cardputer, i think i need another solution like this:
https://www.reddit.com/r/M5Stack/comments/17lwkrd/i_was_disappointed_with_the_number_of_useful_pin/
(Which i'd like to avoid)

bmorcelli, 2024-07-19:
We intend to use the ext I/O from M5Stack to integrate with cc1101, but there's a lot of things to consider to make it work, like:
* Voltage issue,going from 5 to 3,3V
* Mimic SPI communication with the Ext I/O (something that needs a lot of study)

Other thing that should work (easily I think) is programming a ESP32S3 mini to interface with CC1101 through SPI and with Cardputer/StickCs with I2C(Grove port)... It would be an easy port though, but will need to buy an extra mcu ..

I have all these devices at home, but need time to make things happen

eadmaster, 2024-07-21:
I am not very found of this solution because:

- i already have a CC1101 connected via SPI to a regular ESP32 running this [Tasmota fork](https://github.com/arendst/Tasmota/discussions/14343)
- would need an extra module + a lot of wiring, reducing the portability of the whole thing (unless a custom PCB or case is also made)
- need to (partially) rewrite the RcSwitch lib to make use of it. May not work correctly for RAW signal streaming due to the higher latency.

Another solution could be using one of [these modules](http://x311.blogspot.com/2017/10/comparison-of-cheap-rf-modules-with-ask.html), since they only require 1 data pin, you could re-purpose the SDA and SCL pins on the groove connector to drive them directly (e.g G1->TX, G2->RX). I actually have 1 set of these so i might give it a try in the future.

Btw in the meanwhile i'll stick to the Cardputer+[RF433T](https://docs.m5stack.com/en/unit/rf433_t) combo i already have. Since It is able to use the same codes i've captured on Tasmota (tested with [this serial cmd](https://github.com/pr3y/Bruce/issues/64#issuecomment-2231532320)), I am going to add an unofficial protocol to the 
[Subghz File Format](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/file_formats/SubGhzFileFormats.md) to support these codes directly. e.g.:

````
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RcSwitch
Bit: 24
Key: 44 55 33
TE:174
````

This should correspond to `{"Data":"0x445533","Bits":24,"Protocol":1,"Pulse":174,"Repeat":10}'`

bmorcelli, 2024-07-21:
Me neither.. but the only downside of the M5modules and these single pinned modules is that they work only in 433Mhz..

But the thing is, we keep developing for what we have in hand, that are the M5 RF433T and R (that will work with the cheaper ones, regarding the pin setting)

I'm very curious and excited to see your code running in this project..

Ademolght, 2024-07-23:
Can work like this I think 
[image]

eadmaster, 2024-07-23:
> Can work like this I think [...]

In theory it could, but in pratice there are a few caveats to sort out as stated before: different logic voltage, rc libraries expecting to write directly to gpio pins instead if i2s, extra latency due to the i2c bus itself. I've tried bitbanging SPI in the past and it is not an easy task, so i won't be trying this again here.

The easier solution would be having 4 extra gpio pins on the cardputer. According to [this table](https://github.com/m5stack/M5Unified?tab=readme-ov-file#esp32s3-gpio-list) there are actually GPIO16-18, 45, 47, and 48  which are unused, but afaik they are not easily accessible , maybe they could via an hardware mod.

Ademolght, 2024-07-23:
Okay

eadmaster, 2024-07-25:
Prolly this is easier:

1. reconfigure the groove port as a software serial / uart port
2. wire a 2nd microcontroller running this firmware https://github.com/mcore1976/cc1101-tool

No need of bitbanging spi, no voltage shifting required, and much easier protocol.

do you know if 1. can be done on a cardputer?

IncursioHack, 2024-07-25:
I didn't really have time to test much, but come on.

M5Stack's PCA9554 can solve this problem with CC1101, this insight came from an old Everson project that demonstrates the use of PCA9554 with radio frequency devices.

https://www.youtube.com/watch?v=ft_q9qrzepI

With reference to the voltage regulator, a KIA1117PI33 is used to reduce from 5V to 3.3V, I believe that this can be fitted to the CC1101 itself, not generating a cost or a large amount of wires.

https://semiconductors.es/datasheet/KIA1117PI33.html

Another possibility is to get the 3.7V from the battery and bring a wire to the outside area of the Cardputer, this increase in voltage is about 10%, the CC1101 should hold up without problems, but more tests are needed.

https://docs.m5stack.com/en/core/Cardputer

bmorcelli, 2024-07-25:
> reconfigure the groove port as a software serial / uart port
> wire a 2nd microcontroller running this firmware https://github.com/mcore1976/cc1101-tool
> No need of bitbanging spi, no voltage shifting required, and much easier protocol to work with.

This is what I mean by:
> Other thing that should work (easily I think) is programming a ESP32S3 mini to interface with CC1101 through SPI and with Cardputer/StickCs with I2C(Grove port)... It would be an easy port though, but will need to buy an extra mcu ..

But your idea of inteerfacing them by Serial is much better! the MCU is very cheap

eadmaster, 2024-07-26:
quick recap on this:

 - Custom SubGhz was merged yesterday in https://github.com/pr3y/Bruce/pull/82
 - currently only tested with [RcSwitch codes](https://github.com/pr3y/Bruce/issues/66#issuecomment-2241375142), RAW codes are still untested and may not work.
 - only tested by me with a RF433T module, should also work with single-pinned modules like the FS1000A. CC1101 support is still WIP.

[Test build here for the cardputer only](https://github.com/eadmaster/Bruce/releases/tag/1.2.2.wip).

mmjud, 2024-07-27:
Has anyone tried using the HC-11 RF module?

eadmaster, 2024-07-27:
HC-11 and HC-12 cannot be used for custom SubGhz:

> the main purpose of the HC-12 is to create a transparent wireless link between two HC-12's rather than being a generic board. If you are looking to have wireless detectors, switches and other "home automation" gadgets then it's probably not the way to go. 

[(source)](https://www.matrixtsl.com/mmforums/viewtopic.php?t=22996)

Little update for CC1101 support:

 - [this firmware](https://github.com/Stachugit/CatHack) provides CC1101 support for the M5StickCPlus2 only, but it is currently closed source
 - [this library](https://github.com/MajenkoLibraries/SoftSPI) could help bitbanging SPI on PCA9554PW, but as i've said i prefer other solutions not  requiring a library rewrite.

This [uSD card mod](https://www.reddit.com/r/M5Stack/comments/17lwkrd/i_was_disappointed_with_the_number_of_useful_pin/) could also work via [ribbon cables like this](https://it.aliexpress.com/item/1005001928258594.html), as suggested on reddit. So no need to desolder the SD socket, but would need to solder the ribbon cable to the CC1101 module.
I've fould also [these breakout boards](https://www.sparkfun.com/products/9419).

Speaking of ribbon cable mods, the LCD is also using SPI, and 2 devices can share to the same bus via the CS=ChipSelect pin. The hard part here is to make a PCB to split the lines between the 2 modules.

eadmaster, 2024-08-04:
Closing this as it was merged already, open other issues for bugs/improvements /etc

closed 2024-08-04 by eadmaster