BruceDevices/firmware #68: Problème RF Spectrum

softm5, 2024-07-17:
Bonjour,
J'ai un rf433r et lorsque j'essaie le "RF" (Spectrum) sur m5 stick c plus 2 cela ne fonctionne pas.
Ça redémarre le programme Bruce.

bmorcelli, 2024-07-17:
Salut, j'essaie de résoudre ce problème en ce moment... je soupçonne qu'il démarre les mauvaises broches... donc j'espère envoyer un patch pour cela bientôt

Désolé pour la mauvaise traduction de Google Translate

softm5, 2024-07-17:
D'accord merci :)
(Ça à bien traduit)

bmorcelli, 2024-07-19:
Found the bug...

In the last update of arduino framework for esp32 the parameters of the function rmt_config() have changed, and the Macro ESP_ERROR_CHECK() is making it reboot after the error message on Serial...

You'll find the call for this function on line 39 of src/rf.cpp file

Now we can search for a way to fix it..

bmorcelli, 2024-07-22:
Wull be fixed in https://github.com/pr3y/Bruce/pull/71

closed 2024-07-22 by bmorcelli