BruceDevices/firmware #74: Bruce customization

valentin8709, 2024-07-23:
Hi,  
Thank you for your work, it's really great !  
I made a fork to customize and learn. I've fixed some things and added some features.  

[Here is my repo](https://github.com/valentin8709/bruce-extended) and actual state:

*Only tested on M5StickC Plus2.*

**Fixes**
- [x] set & display clock from RTC
- [x] UI bug fixes (like Wifi info drawn on top of the UI boders)
- [x] return to main menu fixed and added when missing
- [x] better shutdown to avoir rolling menu (code from M5 demo)
- [x] default brightness to 25% (little diff but much better battery life)

**Customizations & features**  

**WiFi**  
- [x] conect to default AP (hardcoded ssid / pwd) to avoid typing passphrase on M5StickC Plus 2

**BLE**
- [x] simple BT connection (code from M5 demo)
- [ ] *WIP* BT recon / scanner 
- [ ] test BT CVE (keyboard connection, mic recording etc.)

**IR**
- [x] new IR menu with its icon
- [x] move TV-B-Gone & custom IR here

**FM**
- [x] new FM menu with its icon
- [ ] broadcast music
- [ ] hijack traffic alerts like [I did with a Raspberry Pi](https://github.com/valentin8709/gbadrive)
- [ ] others features with 433 / 868 Mhz modules (not received yet)

**Palnagotchi**
- [x] new Palnagotchi menu with its icon
- [ ] *WIP* adapt code from viniciusbo in order to let M5StickC Plus 2 interacts with Pwnagotchi

**Others**
- [x] QRCodes display (code from Nemo)
- [x] Mic spectrum (code from M5 demo)

I am just a beginner, but anyone can test and take my code if it helps :)

Ademolght, 2024-07-23:
I very like the idea of hijack trafic alert

bmorcelli, 2024-07-24:
Woooow!! I really liked your repo.. 

Feel free to join us on Discord and discuss a Pull Request for your features...

Mark purei to give you dev status

I will test your upgrades to see what need to be changed in order to have it working on the other devices, such as Cardputer (and core in the near future)

closed 2024-08-01 by valentin8709