BruceDevices/firmware #107: port to a headless ESP32-S3 board

eadmaster, 2024-08-02:
In [this branch](https://github.com/eadmaster/Bruce/tree/esp32s3generic) i am trying to port Bruce for use with a generic ESP32-S3 board ("headless"=without LCD and buttons). Currently i'm using [this one](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html).
	
To keep the build as generic as possible, i'm planning to use:

 - a virtual display simulated on a smartphone using [this lib](https://github.com/arpruss/vectordisplayarduino). It  allows to use the touchscreen for input, so no extra buttons are needed.
 - [generic IR modules like these](https://www.keyestudio.com/products/free-shipping-keyestudio-digital-ir-infrared-transmitter-transmit-module-for-arduino) (should be compatible with the ones on M5Stack products, 0 code changes needed)
 - single-pinned modules like the FS1000A for RF (CC1101 should be easy to add in the future via SPI). Still 0 code changes needed for this.
 - builtin wifi and BT (should work without changes)
 - SD card disabled by default, maybe in the future could access storage from the smartphone directly (need a custom app)
 - badusb may need reworking, maybe it's possible to use the 2nd usb port for it (need to investigate)

**So the only parts that needs reworking are: the display, keyboard input, and badusb.**

So far i'm getting the firmware bulding and drawing on my smartphone. Unfortunately the Vectordisplay lib lacks some drawing functions used by Bruce, so it is **not rendering 100% correctly**.
I also have to rewrite the input system to make use of the touchscreen, so this will take a while.

Once i got this working correctly, it should make easier to extend support for other headless ESP32 boards from M5Stack, like the [AtomS3 Lite](https://shop.m5stack.com/products/atoms3-lite-esp32s3-dev-kit), [Capsule Kit](https://shop.m5stack.com/products/m5stack-capsule-kit-w-m5stamps3), and even other generic ESP32 boards.

bmorcelli, 2024-08-02:
Niiice!!!

This morning I will work on the porting to other devices, so I will chage platformio.ini and use flags to allow the features per device...

With it, core devices and CYD will come together

eadmaster, 2024-08-03:
After experimenting with the vectordisplay lib, i've found the easiest route for headless devices should be adding more features for remote control directly to the webUI. 

After checking [your PR here]( https://github.com/pr3y/Bruce/pull/111), i've added back my `esp32-s3-devkitc-1` target, and made [another branch for this solution](https://github.com/eadmaster/Bruce/tree/esp32s3headless).

vectordisplay.h is now just a mock class, and the board autostarts in AP mode.
I've added a new function to send ir files from the webUI:

[image]

Can also be tested with `curl -XPOST "http://bruce.local/ir" -d "filePath=/AC_LG_SX122CL_on.ir"`

Currently WIP:

- not sure how to add a new custom icon to the webui
- add a function to send `sub` and rubbeducky files.
 - move the client-side code of the webui on the flash storage: (1) to save space on the app prtition, (2) for easier development and customization 
 - add [ECM/RNDIS mode](https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/device/usb_dongle/README_EN.md) as alternative to AP mode , so wifi features are still usable

eadmaster, 2024-08-04:
update: i've refactored the  serialcmds module, so that the same commands can also be sent via the webUI. e.g.:

`curl -XPOST "http://bruce.local/cm" -d "cmnd=ir tx NEC 04000000 08000000"`

i've also added 2 new serial cmds:
````
	subghz tx_from_file  plug1_off_key.sub
	ir tx_from_file  LG_AKB72915206_power.ir
````

eadmaster, 2024-08-04:
OT: in the branch i'm working on i've also added another keyboard shortcut for the cardputer only: quick select options using the top row number keys 1-9. Then press again the same number to confirm. Should be a quicker alternative to the usual arrow keys+enter combo.

eadmaster, 2024-08-04:
Quick recap on this:

- for any target, if `HAS_SCREEN`  is not defined, the board will autostart in 'headless' mode, which imply:

- webui and wifi in AP mode started soon after boot
- no menu is rendered and no hardware buttons are checked, the board will only process commands sent via the webui or the serial port
-  not all firmware features are available in this mode. To expose a new one: add a serial cmd, then modify the webui frontend for easier access.

closed 2024-10-22 by bmorcelli