BruceDevices/firmware #113: port tone function to other boards

eadmaster, 2024-08-03:
[current implementation of the `_tone()` function](https://github.com/pr3y/Bruce/commit/6cc077b2612a43929294016dde38521eee592221) won't work with boards that does not have a single pin buzzer, like the cardputer.

Temp workaround: move this constant in the platformio.ini and add a test macro in the code:
````
#define BUZZ_PIN 2
````

Better solution:
add a squarewave generator for boards that have a speaker and [play the waveform with ESP8266Audio](https://github.com/earlephilhower/ESP8266Audio/blob/master/examples/PlayWAVFromFunction/PlayWAVFromFunction.ino)

bmorcelli, 2024-08-04:
True... 

I'll move it to platformio.ini and add a flag on "features" to fix it. Thank you!

eadmaster, 2024-08-06:
closed via https://github.com/pr3y/Bruce/pull/130

closed 2024-08-06 by eadmaster