BruceDevices/firmware #116: 125khz RFID tag reading functionality

geo-tp, 2024-08-03:
It would be nice if Bruce could include RFID125khz tag reading with a Grove RFID UART module
https://wiki.seeedstudio.com/Grove-125KHz_RFID_Reader/

Working Example
```
#include <M5Cardputer.h>
#include <FastLED.h>

#define PIN_LED    21 // Builtin
#define PIN_RX      1
#define PIN_TX      2
#define NUM_LEDS    1

CRGB leds[NUM_LEDS]; // init RGB LED builtin
HardwareSerial RFIDSerial(1);  // UART1

unsigned char buffer[64];  // buffer array for data received over serial port
int count = 0;             // counter for buffer array

M5GFX &Display = M5Cardputer.Display;

void clearBufferArray() {
    for (int i = 0; i < count; i++) {
        buffer[i] = 0;
    }
}

void displayBuffer(unsigned char* buffer, int length) {

    // Reset Screen
    Display.fillRect(0, 67, 240, 70, TFT_BLACK);
    Display.setTextColor(TFT_WHITE);
    Display.setTextSize(1.4);

    // STX
    Display.setCursor(10, 67);
    Display.printf("STX: %02X", buffer[0]);

    // HEX
    Display.setCursor(10, 80);
    int halfLength = (length - 2) / 2;
    for (int i = 1; i < length - 1; i++) {  // data between STX and ETX
        if (i == halfLength + 1) {
            Display.setCursor(10, 93);
        }
        Display.printf("%02X ", buffer[i]);
    }

    // ETX
    Display.setCursor(10, 106);
    Display.printf("ETX: %02X", buffer[length - 1]);

    // ASCII
    M5Cardputer.Display.setTextColor(TFT_ORANGE);
    Display.setTextSize(1.1);
    Display.setCursor(55, 120);
    Display.printf("ASCII: .");
    for (int i = 1; i < length - 1; i++) {
        Display.printf("%c", buffer[i]);
    }
    Display.printf(".");

}
void setup() {
    // UART init
    RFIDSerial.begin(9600, SERIAL_8N1, PIN_RX, PIN_TX);

    // Cardputer Init
    auto cfg = M5.config();
    M5Cardputer.begin(cfg, true);
    Display.setRotation(1);
    Display.setTextColor(TFT_LIGHTGRAY);
    Display.setTextSize(2);
    
    // Display Title
    M5Cardputer.Display.drawRect(10, 10, 220, 40, TFT_LIGHTGRAY);
    M5Cardputer.Display.setTextColor(TFT_LIGHTGRAY);
    M5Cardputer.Display.setCursor(18, 18);
    M5Cardputer.Display.setTextSize(3.2);
    M5Cardputer.Display.printf("RFID 125KHZ"); 
    
    // Dislay Tag infos
    M5Cardputer.Display.setTextSize(1);
    M5Cardputer.Display.setCursor(68,55);
    M5Cardputer.Display.setTextColor(TFT_ORANGE);
    M5Cardputer.Display.printf("Waiting for tags");
    M5Cardputer.Display.setTextColor(TFT_LIGHTGRAY);

    // Builtin LED init
    FastLED.addLeds<WS2812, PIN_LED, GRB>(leds, NUM_LEDS);

}

void loop() {
    if (RFIDSerial.available()) {
        while (RFIDSerial.available()) {  // Read data
            buffer[count++] = RFIDSerial.read();  // Write data into buffer
            if (count == 64) break;
        }
        // Led & Buzzer
        M5Cardputer.Speaker.tone(5000, 20);
        leds[0] = CRGB::Green;
        FastLED.show();
        delay(20);
        FastLED.clear(true);

        displayBuffer(buffer, count); 
        clearBufferArray();
        count = 0;
    }

    delay(200); 

}
```

rennancockles, 2024-08-04:
The rfid module from m5stack does not read 125kHz and I don't have any to test. It's on our roadmap to add the 125kHz functionality, but first we need to get a reader to test it.

rennancockles, 2024-08-14:
Done in this PR https://github.com/pr3y/Bruce/pull/182

closed 2024-08-14 by bmorcelli

geo-tp, 2024-08-16:
I just tried and it works with the grove module.

Thanks.

hahmann, 2024-12-06:
I've been using Bruce for just a couple weeks on a Lily T-Embed CC1101 and I don't have a 125kHz reader but if I try to use the 125kHz functionality selecting it from the menu, the T-embed starts to work very slowly, the SD card won't show up anymore and I'll have to turn-off / turn-on again to fix this behavior. My expectation would be that if the module is not connect, a message will show up stating it instead of needing to reboot, but not sure if this is possible or not. Thank you for the great software you guys developed!

bmorcelli, 2024-12-06:
This lag issue had been fixed in the BetaRelease version.. the error message idk if it is possible to include, but at least it's not lagging anymore