BruceDevices/firmware #119: Bluetooth Jammer

vla-droid, 2024-08-04:
It would be cool to have a bluetooth jammer in the m5. Like you could have a bluetooth speaker connected to your phone playing music, and when you run the thing on the m5, every  bluetooth connection in range would be jammed so the music would stop playing.
https://github.com/alamar808/TheBluetoothJammer
Thanks.

IncursioHack, 2024-08-04:
Despite the super interesting proposal (I have a personal interest in this topic), the example you brought only seemed effective with multiple devices. Another point that I noticed at the end of the text is the chance of success of around 1.5%. 

Out of curiosity, do you have any other projects that you know of that are more effective or more recent with examples and the like? I am also considering that the most current devices have new versions of Bluetooth.

vla-droid, 2024-08-04:
I saw some tiktoks that used an esp32 for this. I'll share my finds.
https://github.com/EmenstaNougat/ESP32-BlueJammer
https://vm.tiktok.com/ZGe7WmvW2/
https://vm.tiktok.com/ZGe7W9294/
I don't know about the antennas but on close range it may work.
Thanks for your answer!

IncursioHack, 2024-08-04:
@vla-droid This example you mentioned now seems to be more functional, assuming the videos are real, I'm afraid the Cardputer/StickC antenna won't do the job.
The examples show two nRF24L01 (I have this board here), it's a shame the code isn't published.

vla-droid, 2024-08-10:
ouch, it would've been a great idea though.
Thanks!!!!

closed 2024-09-07 by IncursioHack