BruceDevices/firmware #120: Version 1.3. Enhancement

Z-SHOCK-E, 2024-08-04:
Finally the settings will be saved. 
I flashed the firmware and it was way better than the previous one.

I just got these ideas that I think will help:

* Moving the "set time" functions of the clock to the "Config".
* Clicking the clock icon displays the time. Without needing be reminded of the time every time it's been clicked.
* The pause function on custom ir is great. It'll be better if it can be resumed too.
* Making that raw sniffer to work without SD card (If possible).
* Bluetooth DDoS attack.

eadmaster, 2024-08-05:
suggestion for more mic graphs here
https://github.com/pr3y/Bruce/issues/125

> BLE badusb. (you can study the code from the portal.hack demo firmware).

Isn't that closed source?

Z-SHOCK-E, 2024-08-06:
Hmm... Unfortunately yes.

Z-SHOCK-E, 2024-08-07:
Rescanning while deauth attack to detect channel hopping.

Z-SHOCK-E, 2024-08-10:
The mic graph on the demo updated way faster. It'll be nice to have it with Bruce too.

closed 2024-09-20 by bmorcelli