BruceDevices/firmware #123: A few requests

edeot, 2024-08-04:
Hello dev team, I'd like to start by saying Bruce is my fav firmware for my cardputer. Great job.

While using your firmware for productivity purposes, I do have a few ideas and requests which I figure couldn't hurt to ask about.

First request is to put the BadUsb function into the main menu with the WiFi, and not in the others menu. It's a powerful tool and I think it is a major use for this firmware along side of the wifi functions. 

Second request, is more of a few ideas. 
Bruce is already so great and I think it has potential to be even more of a Swiss army knife, if we had more functions, like for instance:

-2fa token 

-random password generator (which can automatically be saved into bad usb scripts)

-mouse jigger

-an added partition for data exfil badusb scripts

-travel router (probably not possible)

I understand I have no idea what I'm talking about so these are just some fun suggestions which I think would be awesome to add to the functionality of your awesome firmware. 

Thanks for your time.

And thank you all for your work on this OS

eadmaster, 2024-08-05:
for passwords there is this that would nice to integrate
https://github.com/Joengenduvel/espwv32

eadmaster, 2024-08-18:
little update: i am adding aes encrypted files decoding via openssl-generated files.
The idea is to store a password in each file and decrypt when needed via the file manager.

Basically it is the same as [Passwordstore](https://www.passwordstore.org/), but using symmetric encryption instead, so no need to generate a public+private keypair.

closed 2024-12-07 by bmorcelli