BruceDevices/firmware #133: NFC question

Ademolght, 2024-08-06:
Its possible to load .nfc file if is not , possible add it on Bruce?
Thanks

bmorcelli, 2024-08-06:
I think that renaming them to .rfid it will be read by Bruce..

They look similar to the .rfid we have in this repo..

Ademolght, 2024-08-06:
This Nfc repo can work?
https://github.com/UberGuidoZ/Flipper/tree/main/NFC

Ademolght, 2024-08-06:
But when i see the uberguidoz amiibo repo its 134 pages and when i see the nfc file for Bruce repo its 135 pages

bmorcelli, 2024-08-06:
I think its not a problem... The problem with AMIIBO is that it nees to "lock" the rfid tag.. @rennancockles told me that it isn't beig done yet.. but I think he is the best guy to answer about it

Ademolght, 2024-08-07:
I buy rewritable ntag215 card and write the NFC file on I think I can work

closed 2024-08-17 by Ademolght