BruceDevices/firmware #139: allow storing config.conf in littlefs and avoid EEPROM use

eadmaster, 2024-08-07:
if the SD card is missing store config.conf  in littlefs, completely removing use of EEPROM.
(`bruce.conf` would be a better name)

Also add a setting for the Rf module in use:

  - single-pinned modules ()
  - CC1101 (fixed pinout for each board, defined in platformio.ini)

eadmaster, 2024-08-08:
> if the SD card is missing store config.conf in littlefs, completely removing use of the EEPROM for settings.

@bmorcelli : is it ok if i proceed with this, or are there reasons to keep a copy of the settings in the EEPROM?

bmorcelli, 2024-08-08:
with M5Launcher providing the ability to remap partitions, wuth 2 options that doesn't have SPIFFS (partition type o LittleFS), it is intersting to keep things saved on EEPROM.

Have a copy in LittleFS in intersting too, because there are some firmwares that erase this part of the memory (Volos Watch, is an example)

other thing that we must consider, is the existence of such partition, that caused this issue https://github.com/pr3y/Bruce/issues/142

eadmaster, 2024-08-09:
2fix: i've reworked the [menu here so that it will try to reinit the cc1101 module everytime the user selects it](https://github.com/pr3y/Bruce/blob/b84d2012027aa00e91a977894b1e5c510ccea083/src/core/settings.cpp#L263), instead of trying only once during boot.

However this only works on the 1st try. On the 2nd try the module is reported as connected even if  it is not. It may be a bug of `ELECHOUSE_cc1101.getCC1101()` or something else related to the [init function](https://github.com/pr3y/Bruce/blob/main/src/modules/rf/rf.cpp#L391). If you have any ideas pls post here.

eadmaster, 2024-08-09:
nevermind, i've just found it was an error in the menu logic, it's fixed now and i can send another pr..

closed 2024-08-21 by eadmaster