BruceDevices/firmware #142: Boot loop issue

Tesseract-runner, 2024-08-07:
Hello,
First all, thanks for this very cool software.

I recently downloaded Bruce and launched it with M5Launcher. 
It ran fine the first time, but now, it stays stuck at the booting animation and loops seemingly forever.
Pressing a key at this stage resets the animation.
I have tried:
- rebooting the cardputer
- clearing the FAT
- loading a blank .bin file to clear the SPIFFS
- deleting the firmware on the TF card and replacing it with a new one
All to no avail.
Despite my memory clearing efforts, Bruce has kept my UI color change in the bootup animation.

Has anyone got and idea of what is happening and how I can try to fix it ?

Thank you for your help.

bmorcelli, 2024-08-07:
Try removing the config.conf file from the SD Card, or rename it..

if it solve the problem, please share the file contet with us (remove your Wifi credentials first), probably the issue is there

Tesseract-runner, 2024-08-07:
Thanks, I will try and get back to you when it's done.

Tesseract-runner, 2024-08-07:
Ok, I tried removing the config.conf file but the color config is still there and so is the boot loop.

Tesseract-runner, 2024-08-07:
I just tried without the TF card but the behavior is still the same.

Tesseract-runner, 2024-08-07:
After some troubleshooting, it seems that the SPIFFS is not erased when I load a blank .bin file.
That might explain why the issue is still there when I delete the config file or when I remove the SD card.
When I open the .bin file exported by M5Launcher, I get some garbled but partly readable config text.

bmorcelli, 2024-08-07:
Spiffs will only be erased if you try to write something there... 

You can install OrcaOne (that have something to write on spiffs..

Can you please send us the Serial Monitor Log of the crash, so we'll be able to debug it and provide better guidance

Ademolght, 2024-08-07:
Download bruce directly on m5burner or if is the same problem use the idea of @bmorcelli

Tesseract-runner, 2024-08-07:
I downloaded the firmware here: https://bmorcelli.github.io/M5Stick-Launcher/m5lurner.html
I will try @bmorcelli 's idea and post the results.

bmorcelli, 2024-08-07:
In the Launcher, are you using the default partition scheme?

If you used Doom or Uiflow2, it won't have the Spiffs partition, then we will have problem with Bruce

Tesseract-runner, 2024-08-07:
Installing orca one didn't help.
But when I tried setting the partition scheme to default as suggested by @bmorcelli, it worked !
That's weird, because I never touched this setting before.

Anyway, thank you all for you help !

closed 2024-08-07 by Tesseract-runner

HabobB6, 2026-02-28:
I got the stack too after i upload bruce on my custom board