BruceDevices/firmware #143: wifi password cracker?

jkrowtheh4cker, 2024-08-07:
could you add some sort of brute force wifi password cracker capable of operating without an SD card?

7h30th3r0n3, 2024-08-07:
Password cracking need to use a lot of calculations power that the cardputer/stick don't have, even if possible it should be really really slow, like 1 try per second and I'm optimist, so it's better to crack it with a graphic card or online by bsending it to a service designed for this

closed 2024-08-08 by bmorcelli

valentin8709, 2024-08-08:
Just a little idea : in the future, Bruce would be able to capture handshakes from raw sniffing.
Handshakes could be cracked on a computer, but Bruce *could* be able to :
- connect to BT / WiFi hotspot for Internet
- send handshakes to OnlineHashCrack or WPA-Sec websites in case of easy / known password
- test WiFi vulnerabilities that does not need calculations, perhaps like KRACK, FragAttaks, PMKID attak, WPS vulns etc.
- resend newly captured handshakes to OnlineHashCrack with successfull vulnerability exploitation

Ideas come from Pwnagotchi, which runs on a RPi Zero and have more calculation capabilities.
I would love trying to develop stuff like that as soon as I have time to do it :)

jkrowtheh4cker, 2024-08-11:
@7h30th3r0n3 could we just make it save the hashes to littleFS and add a hash cracker to the web UI, so you could just get on webUI and click a button to automatically crack the hashes that the cardputer found?

7h30th3r0n3, 2024-08-11:
> @7h30th3r0n3 could we just make it save the hashes to littleFS and add a hash cracker to the web UI, so you could just get on webUI and click a button to automatically crack the hashes that the cardputer found?

We'll make a hash cracker inside a web browser will be really ineffective while you can just download it and cracking it with well know cracker on a pc like hashcat or airacrack, remember that WiFi password cracking need really big ressources, any hash/s gain is a save of time to wait, the best way would be was @valentin8709 say send it to onlinehashcrack or using your own server with a big RTX that resend the cracked hash to the Cardputer

eadmaster, 2024-08-12:
related: https://github.com/pr3y/Bruce/issues/131

jkrowtheh4cker, 2024-08-12:
yeah... for cracking hashes I mostly use kali linux tools like hashcat.