BruceDevices/firmware #146: AP not found

loewal, 2024-08-08:
AP mode gives me an address of 172.0.0.2
ping is ok, but http://172.0.0.1 gives dns error...
bruce.local finds nothing.
Any idea?

bmorcelli, 2024-08-08:
Are you accessing the wireless network in a computer connected into wired network?

If you're accessing like this, I would suggest disconnect from the wires network to properly connect..

If you are accessing from your smartphone, turn off GSM internet, because dns table you're using will not point to the bruce device

loewal, 2024-08-08:
My iMac is wirelessly connected to the Internet through my Airport Extreme router.
I connect to "bruce", enter password and now my iMac has ip-address 172.0.0.2
Ping in terminal if fine.
http://172.0.0.1 gives dns-error...
Normally those settings are no problem.....

loewal, 2024-08-08:
Some data:
en1: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=6460<TSO4,TSO6,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
	ether 36:92:c1:49:21:3a
	inet6 fe80::105f:bcb1:721a:f65f%en1 prefixlen 64 secured scopeid 0xc
	inet 172.0.0.2 netmask 0xffffff00 broadcast 172.0.0.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
	
	/Users/Loewal/Downloads> ping 172.0.0.1
PING 172.0.0.1 (172.0.0.1): 56 data bytes
64 bytes from 172.0.0.1: icmp_seq=0 ttl=64 time=14.177 ms
64 bytes from 172.0.0.1: icmp_seq=1 ttl=64 time=8.322 ms
64 bytes from 172.0.0.1: icmp_seq=2 ttl=64 time=10.841 ms
64 bytes from 172.0.0.1: icmp_seq=3 ttl=64 time=19.042 ms
64 bytes from 172.0.0.1: icmp_seq=4 ttl=64 time=16.361 ms
64 bytes from 172.0.0.1: icmp_seq=5 ttl=64 time=7.552 ms
64 bytes from 172.0.0.1: icmp_seq=6 ttl=64 time=3.182 ms

[image]
[image]

loewal, 2024-08-08:
Also tried to reach on iPhone. Same result.

loewal, 2024-08-08:
Tried AP mode on M5launcher: success....so...strange

IncursioHack, 2024-08-08:
Version: Bruce 1.3.1 - Oficial - M5Burner
Version: Windows 10 10.0.19045
Browser: Chrome 127.0.6533.89
08/08/24 - 09:55

Menu > Others > WebUI > AP Mode
IP: 172.0.0.1
URL: http://172.0.0.1
URL: http://bruce.local

[image]
[image]
[image]
[image]

It seems normal to me.

IncursioHack, 2024-08-12:
@loewal 
If possible, delete the current config.conf file from your device, save the version of that URL ( https://github.com/pr3y/Bruce/actions/runs/10342227214 ), and generate a new file. After that, perform the test again and let us know if the error still occurs.
Unfortunately I don't have a MacOS to simulate your mistake. With reference to Windows I had no problem (evidence above).

bmorcelli, 2024-08-23:
Brucenet wifi's password is "brucenet"

It will be written on screen

closed 2024-08-23 by bmorcelli