BruceDevices/firmware #159: conf.conf How to connect to the saved wifi. It keeps asking for the password?

Z-SHOCK-E, 2024-08-10:
After I saved the network creds in the conf file it still asks for the wifi password?

IncursioHack, 2024-08-12:
@Z-SHOCK-E , If possible, delete the current config.conf file from your device, save the version of that URL ( https://github.com/pr3y/Bruce/actions/runs/10342227214 ), and generate a new file. After that, make the changes you think are relevant.
Because you made a manual change, I can't reproduce your error.

bmorcelli, 2024-08-13:
It is working on with SD for now, but config file being saved  on LittleFs

Need take the SD checks here https://github.com/pr3y/Bruce/blob/main/src%2Fcore%2Fwifi_common.cpp#L26 and here 
https://github.com/pr3y/Bruce/blob/main/src%2Fcore%2Fwifi_common.cpp#L61

bmorcelli, 2024-08-14:
Fixed here https://github.com/pr3y/Bruce/pull/183

closed 2024-08-14 by bmorcelli