BruceDevices/firmware #162: RD module support

geo-tp, 2024-08-10:
This RF module seems really close to the capabilities of the flipper one  (433/470/868/915 with Grove port and UART interface).

https://www.seeedstudio.com/Grove-Serial-RF-Pro.html

Could it be supported by Bruce ?

eadmaster, 2024-08-10:
notice that this supports FSK modulation only, not ASK/OOK which is the most commonly used.

Iinm this is akin to the [HC-11/HC-12 modules](https://github.com/pr3y/Bruce/wiki/RF) -> not meant for sniffing/replaying, but to get a simple transparent uart<->rf bridge between 2 boards.

Z-SHOCK-E, 2024-08-20:
There are some IO expansion boards from m5stack that I hope will solve the lack of IO pins on the cardputer for connecting the cc1101 module

https://www.aliexpress.com/item/1005004638915793.html?spm=a2g0o.store_pc_allItems_or_groupList.0.0.17fc6325s5YuVH&pdp_npi=4%40dis%21LKR%21LKR%202%2C372.81%21LKR%202%2C372.81%21%21%217.95%217.95%21%4021010c9c17241472693227741e896c%2112000029933355066%21sh%21LK%212397466382%21X

https://www.aliexpress.com/item/1005004031007705.html?spm=a2g0o.store_pc_allItems_or_groupList.0.0.17fc6325s5YuVH&pdp_npi=4%40dis%21LKR%21LKR%202%2C372.81%21LKR%202%2C372.81%21%21%217.95%217.95%21%4021010c9c17241472693227741e896c%2112000027801166398%21sh%21LK%212397466382%21X

https://www.aliexpress.com/item/1005003297346245.html?spm=a2g0o.store_pc_allItems_or_groupList.0.0.7ee66325hczy80&pdp_npi=4%40dis%21LKR%21LKR%201%2C044.63%21LKR%201%2C044.63%21%21%213.50%213.50%21%4021010c9c17241472433036473e896c%2112000025077696818%21sh%21LK%212397466382%21X

I did't look into the chipsets used in them though.

bmorcelli, 2024-08-20:
These are I2C extenders, their pins can't be controlled individually 

The only ones available for this task would be https://docs.m5stack.com/en/unit/extio2

But you need to "mimic" the SPI comunication by switching/reading the pins individually.. and this is the tricky part

Z-SHOCK-E, 2024-08-20:
Ah ok. So.........are there any example code out there to mimic the SPI communication.

Z-SHOCK-E, 2024-08-20:
Or can we use a seperate (small in size) and program it so that the cc1101 module can be accessed using the serial console which only requires 4 pins.

Check this out https://github.com/mcore1976/cc1101-tool

eadmaster, 2024-08-20:
> https://github.com/mcore1976/cc1101-tool

As alternative, we may have a 2nd headless board running Bruce as well and talking to the 1st one via [serial cmds](https://github.com/pr3y/Bruce/wiki/Serial).

So e.g. when you want to send a .sub file, the main board will send its contents to the 2nd for transmitting via the cc1101.

Same logic could be used for badusb scripts that the 1st board may not be able to run directly due to lack of usb hid.

Z-SHOCK-E, 2024-08-21:
Hmm....Yea.
but running another esp32 board will be a little wasteful because
*It'll use more power compared to the pro mini (I assume)
* It'll cost more
But I found this tiny esp32  board that looked promising.
( I know that coding another board is a headache rather than using the already developed code and making it communicate via serial cmds )
https://cutt.ly/tiny32

bmorcelli, 2024-08-21:
This is one of the candidates to use as a side MCU to run all the other hardwares pieces..

The problem with the C3 is that it only has one core and can't run BadUSB... The S3 Super mini or S3 Zero have the same size, but 2 cores and can run as HID, so it'll be better for stickCs integration..

eadmaster, 2024-08-21:
> It'll use more power compared to the pro mini (I assume)

You could save some power forcing the board into lightsleep after each serial command, and waking up when there is a new command:

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html#uart-wakeup-light-sleep-only

Power consumption in light-sleep is around 0.8 mA. ([source](https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/#esp32-light-sleep))

Jawen33, 2024-08-21:
Hi, I have a LoRaE220--433, how ton use with this firmware please?
[image]

IncursioHack, 2024-08-21:
> Hi, I have a LoRaE220--433, how to use with this firmware please? ![17242613962601467344266339639998](https://private-user-images.githubusercontent.com/21220398/360019413-924c9b6d-6c86-44bc-b02d-3c80c1a2511d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQyNjE3NzUsIm5iZiI6MTcyNDI2MTQ3NSwicGF0aCI6Ii8yMTIyMDM5OC8zNjAwMTk0MTMtOTI0YzliNmQtNmM4Ni00NGJjLWIwMmQtM2M4MGMxYTI1MTFkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIxVDE3MzExNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlOWJkNWM0MzY5ODlhNjlhMzA5OTVhOTBmY2Y3NGExYTg4YzY2ZTMyZTQ1Mjc4OTE3MmUxMTkzZjc0ZWVlNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3jnFPNTy8zboeD8icn-KyHU-kuNYBulQf8G-0nFSA84)

LoRa 433 and RF433 are different and have different functions, we do not have LoRa implementation in this firmware.
I suggest looking for a firmware called "Cardputer LoRaChat", there you should find something.

Jawen33, 2024-08-21:
Thanks for your answer, I will look for it. I hope there are all the features like Bruce 🍀

Jawen33, 2024-08-21:
[image]

Z-SHOCK-E, 2024-08-23:
I'll buy this s3 board https://cutt.ly/zero32. Waiting for the firmware to be released for the zero mini s3

eadmaster, 2024-08-23:
One thing you may want to check is if the board has support for power input via the 5V/Vin pin, so it could be powered directly from the groove connector.

I was able to do that with the [board i have currently, but it required a little mod](https://forum.arduino.cc/t/chinese-esp32-s3-5v-pin-warning/1192758).

Having some Psram would also help to transfer and cache files with this lib i'm currently using:
https://github.com/tobozo/ESP32-PsRamFS

eadmaster, 2024-08-23:
note that the current `esp32-s3-devkitc-1` env i have added platformio.ini should also be compatible with these ESP32-S3 Mini boards (but i don't have one  to test currently):
https://community.platformio.org/t/need-help-add-esp32-s3-mini-in-platform-io-in-vs-code/26041

you may need to change some unexposed pin numbers, and i also like to add psram support that some of these boards have, because it helps with the coding.

As alternative for a small board, i have one of these which comes with some psram and have more usable pins: https://www.wemos.cc/en/latest/s2/s2_mini.html

eadmaster, 2024-08-25:
> One thing you may want to check is if the board has support for power input via the 5V/Vin pin, so it could be powered directly from the groove connector.

I did some initial testing for this, and it does not seem to be so reliable as i thought:

> Are you trying to supply the second ESP32 through the 3.3V pin of the first one? This will not work. The ESP32 produces current-spikes 300 mA high.

([source](https://forum.arduino.cc/t/cant-get-uart-connection-between-two-esp32s-solved/968366/7))

Even with a separate power source it is still not working as expected, so i think i'm stuck atm.

If you want to take a try , the code is here:
https://github.com/eadmaster/Bruce/tree/headlessfixes

(There is a new menu entry in RF->Config->RF Module)