BruceDevices/firmware #164: [Bug Report] RF Modules is freezing Bruce

IncursioHack, 2024-08-10:
[Test Bruce] 1.3.1  /248 - Dev -  https://github.com/pr3y/Bruce/actions/runs/10332927025
[Version/Artifact] Bruce_m5stack-cardputer_248
[Datetime] 08/10/2024 - 13:30 [GMT-3]
[Device] Cardputer

[Bug Report]

RF Modules
When selecting Config > RF Module > CC1101 on SPI on the Cardputer (without having a CC1101 connected), Bruce freezes.

[13:46:16]Selecionado RF Module
[13:46:19]Selecionado CC1101 on SPI
[13:46:20][ 34293][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420a2c90 arg=0x3fc9c52c

eadmaster, 2024-08-10:
this is what you should see, maybe there is a conflict with another setting that is causing the freezing:

[image]

try the new `factory_reset` cmd i've just pushed: https://github.com/pr3y/Bruce/pull/160/commits/3999ddcdf2818ac54036ef288c35eb24ce64a997
(will erase your current settings)

IncursioHack, 2024-08-10:
[image]

I deleted the config.conf file via WebUI. But Bruce continues to freeze when selecting "CC1101 on SPI"

[15:02:45][   498][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface():[  8892][E][vfs_api.cpp:113] exists(): File system is not mounted
[15:02:45][  8899][E][vfs_api.cpp:105] open(): /littlefs/boot.wav does not exist, no permits for creation
[15:02:45][  9135][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420a2c90 arg=0x3fc9c52c
[15:02:45][  9167][I][settings.cpp:710] getConfigs(): getConfigs: config.conf exists
[15:02:45][  9187][I][settings.cpp:722] getConfigs(): getConfigs: deserialized correctly
[15:02:45][  9194][I][settings.cpp:742] getConfigs(): Brightness: 100
[15:02:45][  9200][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[15:02:45][  9210][I][settings.cpp:764] getConfigs(): Wrote new conf to EEPROM
[15:02:45][  9216][I][settings.cpp:766] getConfigs(): Using config.conf setup file
[15:02:45]End of Config
[15:02:55][ 19108][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[15:03:00][ 24108][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
[15:03:05][ 29150][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[15:03:22]Selecionado RF Module
[15:03:25]Selecionado CC1101 on SPI
[15:03:25][ 48945][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420a2c90 arg=0x3fc9c52c

IncursioHack, 2024-08-11:
[Test Bruce] 1.3.1 /249 - Dev - https://github.com/pr3y/Bruce/actions/runs/10333329029
[Version/Artifact] Bruce_m5stack-cardputer_249
[Datetime] 08/10/2024 - 23:05 [GMT-3]
[Device] Cardputer

It was tested on the version of PR #160 / Actions #249.

On the Cardputer with or without the RF433R module connected to the Grove port, the CC1101 option does not appear in the Config>RF Module Menu, therefore, as the option is not displayed, the error no longer occurs.

Maybe we can close this issue and keep it archived so we can be notified about the future implementation of the CC1101 on the Cardputer with SD Card Sniffer.

Thanks for your help, I think I should mention that this part in Config has been resolved.

eadmaster, 2024-08-11:
i was able to reproduce the issue, it seems something SPI-related:

````
[15:03:25][ 48945][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420a2c90 arg=0x3fc9c52c
````

related
https://github.com/greiman/SdFat/issues/462

bmorcelli, 2024-08-11:
In our case we are not sharing tft_eSPI bus with SD.. but sharing SD bus with CC1101.. 

TFT_eSPI tuns in a separate bus for Cardputer (Different from Core Devices that share Sd with Tft)

But the reference is nice to think about the solution..

closed 2024-08-23 by bmorcelli