BruceDevices/firmware #165: [Bug Report] RAW Sniffer rebooting Bruce.

IncursioHack, 2024-08-10:
[Test Bruce] 1.3.1  /248 - Dev -  https://github.com/pr3y/Bruce/actions/runs/10332927025
[Version/Artifact] Bruce_m5stack-cardputer_248
[Datetime] 08/10/2024 - 13:30 [GMT-3]
[Device] Cardputer

 [Bug Report]

Raw Sniffer

When using RAW Sniffer in Wifi > Raw Sniffer, the message "Sniffer Started" appears and does not display any feedback until the ENTER key is pressed. When pressed, a counter of captured packets is displayed.
When exiting and returning to RAW Sniffer, Cardputer restarts itself.
Log below:

[13:50:39][   498][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface():CAP/raw_4.pcap saved!
[13:50:39]==================
[13:50:39]opened: /BrucePCAP/raw_5.pcap
[13:50:39]==================
[13:50:39]/BrucePCAP/raw_5.pcap saved!
[13:50:39]==================
[13:50:39]opened: /BrucePCAP/raw_6.pcap
[13:50:39]/BrucePCAP/raw_6.pcap saved!
[13:50:39]==================
[13:50:39]opened: /BrucePCAP/raw_7.pcap
[13:50:39]Selecionado Raw Sniffer
[13:50:39]
[13:50:39][ 78544][E][vfs_api.cpp:105] open(): /sd/BrucePCAP/raw_8.pcap does not exist, no permits for creation
[13:50:39]opened: /BrucePCAP/raw_8.pcap
[13:50:39]ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40381c7c
[13:50:39]file: "src/modules/wifi/sniffer.cpp" line 194
[13:50:39]func: void sniffer_setup()
[13:50:39]expression: esp_event_loop_create_default()
[13:50:39]
[13:50:39]abort() was called at PC 0x40381c7f on core 1
[13:50:39]
[13:50:39]
[13:50:39]Backtrace: 0x40378746:0x3fcebba0 0x40381c89:0x3fcebbc0 0x40387d8d:0x3fcebbe0 0x40381c7f:0x3fcebc60 0x42038753:0x3fcebc80 0x4200547b:0x3fcebda0 0x420042d6:0x3fcebdc0 0x4200529d:0x3fcebde0 0x42007e9a:0x3fcebe50 0x42007f4d:0x3fcebf40 0x4201ef6f:0x3fcebf60 0x420a5240:0x3fcec000

IncursioHack, 2024-08-10:
As an additional feature, as it is the same function, when pressing enter it ends up generating a double click and saving 2 files at once. I suggest inserting a delay of around 1 second or more.

bmorcelli, 2024-08-14:
Fixed here https://github.com/pr3y/Bruce/pull/183

closed 2024-08-14 by bmorcelli