BruceDevices/firmware #166: [Bug Report] TV Be Gone freezes

andortizg, 2024-08-10:
[Test Bruce] 1.3.1 /248 - Dev -  https://github.com/pr3y/Bruce/actions/runs/10332927025
[Version/Artifact] Bruce_m5stack-cardputer_248
[Datetime] 08/10/2024 - 13:30 [GMT-3]
[Device] Cardputer

[Bug Report]

It is not possible to choose region in the start menu. It switches to "running" (without pressing any key) and Bruce freezes while “running” appears on the display.

IncursioHack, 2024-08-10:
Here it worked for me.

Check:

1 - The version of the Bruce build you are using and test again. ( https://github.com/pr3y/Bruce/actions/runs/10332927025 )

2 - Do you have any module installed?

3 - Was any configuration made in IR TX Pin and IR RX Pin in the Settings menu?

4 - If it hasn't resolved the problem, try making a backup and deleting the config.conf file from the SD card.

https://github.com/user-attachments/assets/2df36aa1-e336-44cb-b979-82e05d177359

andortizg, 2024-08-11:
Deleting the config.conf file worked like a charm.

Thanks!

closed 2024-08-12 by IncursioHack