BruceDevices/firmware #170: [Bug Report] IR Remote module with IR Read does not read infrared controls.

IncursioHack, 2024-08-10:
[Test Bruce] 1.3.1 /248 - Dev - https://github.com/pr3y/Bruce/actions/runs/10332927025
[Version/Artifact] Bruce_m5stack-cardputer_248
[Datetime] 08/10/2024 - 16:00 [GMT-3]
[Device] Cardputer

[Bug Report]
IR Read

When inserting the M5Stack IR Remote Module into the Cardputer's GROVE port, select the Config > IR Rx Pin > M5 IR Mod option, on the Config screen the M5 IR Mod and Grove Y options are in blue.

When selecting the IR > IR Read option, Bruce normally displays the IR reading screen, however when trying to point an LG remote control (AKB73715682) and press some buttons, it does not perform any reading, it was also tested with a remote control. LG air conditioning (AKB75215403) and also did not show the screenshot on the screen.

Signals Captured: 0
Waiting for signals...

[16:00:53][  8891][E][vfs_api.cpp:113] exists(): File system is not mounteEnd of Config
[16:00:56][ 34356][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[16:01:01][ 39356][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0
[16:01:05][ 43716][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[16:01:08]Selecionado IR Read
[16:01:18][ 56958][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 164
[16:01:23][ 61958][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 0

IncursioHack, 2024-08-10:
As information, TV-B-Gone and IR Custom worked perfectly both with the IR Remote module and without, using the Cardputer's native IR, with the appropriate settings in the Config > IR Tx Pin menu.

bmorcelli, 2024-08-10:
> When inserting the M5Stack IR Remote Module into the Cardputer's GROVE port, select the Config > IR Rx Pin > M5 IR Mod option, on the Config screen the M5 IR Mod and Grove Y options are in blue.

Both are in blue because it uses the same pin.. I've left the 2 options for the same pin to simplify the choose process (if you don't have the IR Module, is nice to have the other for reference)

IncursioHack, 2024-08-11:
I discovered the problem, when selecting the "M5 IR Mod" on both the IR TX Pin and the IR RX Pin, it is selecting the same "GROVE Y" pin, where one should be Grove Y and the other Grove X. You can notice this on the Config screen and also in the "config.conf" file.

I think it needs to be adjusted in the code.

[image]

IncursioHack, 2024-08-11:
When I manually changed the IR RX Pin to Grove W, it worked and I was able to capture the signals from the controls.

I believe that the adjustment is only in the association of the M5 IR Mod in the IR RX Pin option, where it must select the Grove W.

bmorcelli, 2024-08-11:
https://github.com/pr3y/Bruce/blob/038da5a64167ca8a1ba5475799edc2ebbfdae8e7/src/core/settings.cpp#L636

This line must be changed to RF_RX_PINS

IncursioHack, 2024-08-12:
[Test Bruce] 1.3.1 /254 - Dev - https://github.com/pr3y/Bruce/actions/runs/10342227214
[Version/Artifact] Bruce_m5stack-cardputer_254
[Datetime] 08/12/2024 - 06:55 [GMT-3]
[Device] Cardputer

The bug has been fixed in the version mentioned above.

I thank you for your commitment @bmorcelli , you are amazing as always.

closed 2024-08-12 by IncursioHack