BruceDevices/firmware #172: Deauth not working

jkrowtheh4cker, 2024-08-10:
deauth works in 1.3 but not in 1.3.1

IncursioHack, 2024-08-10:
Could you perform a test on this version and tell me if the problem still occurs?
Taking advantage, could you also test Deauth+Clone and Flood?

https://github.com/pr3y/Bruce/actions/runs/10332927025

jkrowtheh4cker, 2024-08-11:
how do I use that version (sorry, new to github)

IncursioHack, 2024-08-11:
Download this file and burn/install the firmware using the website https://web.esphome.io/

https://github.com/pr3y/Bruce/actions/runs/10332927025/artifacts/1798049218

jkrowtheh4cker, 2024-08-11:
seems like bmorcelli fixed this...

IncursioHack, 2024-08-12:
[Test Bruce] 1.3.1 /248 - Dev - https://github.com/pr3y/Bruce/actions/runs/10332927025
[Version/Artifact] Bruce_m5stack-cardputer_248
[Datetime] 08/12/2024
[Device] Cardputer

The test was carried out with Deauth (Wifi > Wifi Atks > "Select Wifi" > Deauth), my cell phone (Android - Poco X3 Pro) was used as a router in Portable Hotspot mode, providing internet/network to a Dell Notebook with Windows 11, about 5 to 7 seconds of attack the notebook suffered deauthentication and could no longer connect to the wifi network created by my cell phone. Therefore, I understand that the problem has been solved.

@bmorcelli 

[image]

closed 2024-08-12 by pr3y