BruceDevices/firmware #178: AI?

jkrowtheh4cker, 2024-08-12:
could we add an AI chatbot like the cardputer Gemini that @vanshksingh made
just a thought

vanshksingh, 2024-08-12:
Interesting, I mean I can make a pull request to this project with that added would that be cool ?

bmorcelli, 2024-08-12:
I don't think it adds something to the project scope..

If it comes to bruce, Can be set as a optional feature for those who wants it.. but I fear it'll fill out IRAM table that could be used to a more valuable thing

jkrowtheh4cker, 2024-08-12:
yeah... I forgot about limited IRAM
do you think that the next update will fix the deauther problem / add more deauther features

closed 2024-08-14 by bmorcelli

jkrowtheh4cker, 2024-08-15:
I noticed that on m5launcher/m5burner you can't get the new 1.3.1 with fixes.
do you know why this might be happening?

bmorcelli, 2024-08-15:
It wasn't posted there, probably..

jkrowtheh4cker, 2024-08-17:
I think it might work if you post it to m5launcher/m5burner as 1.3.2 so it won't install the old version