BruceDevices/firmware #190: M5stickc plus 2. Reboots when pasting a copied file

Z-SHOCK-E, 2024-08-18:
When I try to copy and  paste a .ir file on another folder Bruce reboots without transferring the file.

Ademolght, 2024-08-18:
me too with the cardputer

bmorcelli, 2024-08-18:
This problem is related to the size of the buffer here https://github.com/pr3y/Bruce/blob/d05b1c05e28c6c049a2d8872e2dba74220d22e8b/src/core/sd_functions.cpp#L137 and here https://github.com/pr3y/Bruce/blob/d05b1c05e28c6c049a2d8872e2dba74220d22e8b/src/core/sd_functions.cpp#L212

Maybe using this global buffer https://github.com/pr3y/Bruce/blob/d05b1c05e28c6c049a2d8872e2dba74220d22e8b/src/core/globals.h#L94 can solve the pr

Z-SHOCK-E, 2024-08-19:
Ok Thanks.

bmorcelli, 2024-08-23:
Fixed here https://github.com/pr3y/Bruce/pull/194

closed 2024-08-23 by bmorcelli