BruceDevices/firmware #210: emulate usb composite device on capable boards (mouse+pendrive+keyboard)

eadmaster, 2024-08-22:
[example code here](https://github.com/tobozo/WUD-Ducky/blob/main/WUD-Ducky.ino), some interesting features to add:

- mouse support for badusb/rubberducky scripts
- mass storage mode as alternative to the webui for file transfers

bmorcelli, 2024-08-23:
https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard

Saving the library here to sum it with the USB we already use, to save space with common key arrays

eadmaster, 2024-10-25:
just found another example to try, this one also includes a serial port that is currently lost when switching to USB HID mode:
https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/CompositeDevice/CompositeDevice.ino