BruceDevices/firmware #215: Recommend Top Features

FernandoJose2, 2024-08-24:
First of all, thank you very much to all Contributors.

I think Bruce will be more useful when he has these features.

BadKB
We can use RubberDucky script on Bluetooth for mobile phones and computers.

Bluetooth Remote Control
Remotely control computer, mobile phone, camera and other devices with Bluetooth.

BLE FindMy Location Beacon
Use your Cardputer as Airtag and track on Apple, Samsung and Tile location networks.

IncursioHack, 2024-08-24:
Hi FernandoJose2, how are you? I hope well :)

The last point you mentioned, I believe Openhaystack should do this, could you check and test it?

https://github.com/pr3y/Bruce/wiki/Others#openhaystack

FernandoJose2, 2024-08-25:
Hi IncursioHack, Thanks man. I'm good I hope you are good too.

You're right, it's the same feature. I haven't tried it before.

I wonder if the other features have been added or will be eliminated.

IncursioHack, 2024-08-25:
BadKB 
Has already been mentioned in a conversation, but without a preview.

Bluetooth Remote Control
There is something similar to this in other firmware, if you want to check it out.
https://github.com/geo-tp/Bluetooth-Keyboard-Mouse-Emulator/tree/main

bmorcelli, 2024-09-20:
Let's centralize BLE suggestions here https://github.com/pr3y/Bruce/issues/239

This issue is being closed.

closed 2024-09-20 by bmorcelli