BruceDevices/firmware #221: MIMA Mod

FernandoJose2, 2024-08-26:
WiFi's features is very good but If there were the middle attack options, it would be legendary. Would really love seeing these.

IncursioHack, 2024-08-26:
Hi Fernando, so the ESP32 has several limitations in both code and processing (CPU, RAM and ROM).
If you could show us some example code in ESP32 and which Man in the Middle attack you are talking about (I say this because there are several), it would help us better understand what you are thinking about and whether or not it is possible to implement it.

IncursioHack, 2024-08-26:
If you want to discuss an idea, I understand that Bruce's Discord is the best way, so we leave Github issues for cases where we manage to mature something or for bugs in general.

https://discord.com/invite/WJ9XF9czVT

FernandoJose2, 2024-08-26:
I don' use Discord man, but i will join for you asap. It would be nice to track the website addresses that users on the WiFi network you are connected to enter from their browsers.

IncursioHack, 2024-08-27:
I didn't find any documentation on using an ESP32 as a network sniffer to capture http communication. One option for this is the Raspberry Pi Zero2W with Wireshark. Today we have RAW Capture for Wifi in general.

jkrowtheh4cker, 2024-08-28:
the real question is if MIMA with the network sniffing thing is possible though...
it would be awesome

FernandoJose2, 2024-08-28:
> the real question is if MIMA with the network sniffing thing is possible though... it would be awesome

MIMA with the network sniffing is very simple. This is already available on github. I'm talking about redirecting the addresses in the victim's browser to our fake addresses. I say that firmware should manage network traffic.

jkrowtheh4cker, 2024-08-29:
do you mean redirect to a fake google login or something? it would be funny if we added an option to redirect them to the Wikipedia page about hacking

FernandoJose2, 2024-08-29:
> do you mean redirect to a fake google login or something? it would be funny if we added an option to redirect them to the Wikipedia page about hacking

Of course. You can direct them to any fake pages. This is a most commonly used attack method.

jkrowtheh4cker, 2024-08-31:
let's do it, this would be a great feature!

jkrowtheh4cker, 2024-09-05:
is this issue just abandoned or something, it feels a little dead