BruceDevices/firmware #239: Feature request  - BlueDucky (BLE) / MouseJacking

pwunkz14, 2024-09-02:
BlueDucky - Use of the exploiting 0-click Android Bluetooth vulnerability to inject keystrokes without pairing ( all android under 10 version are vulnerable)

MouseJacking  - Wireless Mouse and Keyboard Vulnerability

Niximkk, 2024-09-03:
> inject keystrokes without pairing

is that a thing??? 👀

IncursioHack, 2024-09-03:
> > inject keystrokes without pairing
> 
> is that a thing??? 👀

I believe this is the most complex thing to do, most cell phones, even the cheapest ones, have an Android version above 10. Maybe it's worth checking if the "Android Smart TV Box" has this vulnerability, there are more chances of them being on Android 9 .

> MouseJacking - Wireless Mouse and Keyboard Vulnerability

This is a case that we have been discussing for a few days, if there is any code for ESP32 available, send it to us to evaluate.

Niximkk, 2024-09-03:
> there are more chances of them being on Android 9

Mine is on android 10 with a Feb 5 2023 security patch

Petitoursbr1, 2024-09-03:
Could this code be adapted for Bruce? [https://github.com/dnatividade/Arduino_mousejack](https://github.com/dnatividade/Arduino_mousejack)

pwunkz14, 2024-09-03:
Blue ducky code
https://github.com/pentestfunctions/BlueDucky

samallx, 2025-04-02:
> BlueDucky - Use of the exploiting 0-click Android Bluetooth vulnerability to inject keystrokes without pairing ( all android under 10 version are vulnerable)
> 
> MouseJacking - Wireless Mouse and Keyboard Vulnerability

https://github.com/insecurityofthings/uC_mousejack/tree/master
this is a project made for for feather dev board, but works also on esp32. This is a library. Paste and play