BruceDevices/firmware #242: ARP poisoning - possible or not on an ESP32?

LZ2DMV, 2024-09-03:
My research shows mixed results on wheter or not sending ARP replies is possible on an ESP32.

On one hand, Bruce already uses IwIP to send ARP requests as part of the host scan feature, but a from a quick glimpse at IwIP's documentation, it seems that there is no function for ARP replies, which makes sense for a non-offensive use of the library - ARP replies are handled by the ESP core.

However, is it possible to inject raw Ethernet frames with ESP32?

If it is, then implementing ARP poisoning for Bruce should be a matter of finding an example of an ARP reply frame (constructing it as per the RFC or using some Scapy example) and in-place editing of the source and target IPs.

IncursioHack, 2024-09-03:
> My research shows mixed results on wheter or not sending ARP replies is possible on an ESP32.
> 
> On the one hand, Bruce already uses IwIP to send ARP requests as part of the host scan feature, but from a quick glimpse at IwIP's documentation, it seems that there is no function for ARP replies, which makes sense for an non-offensive use of the library - ARP replies are handled by the ESP core.
> 
> However, is it possible to inject raw Ethernet frames with ESP32?
> 
> If it is, then implementing ARP poisoning for Bruce should be a matter of finding an example of an ARP reply frame (constructing it as per the RFC or using some Scapy example) and in-place editing of the source and target IPs.

In a funny way I spent the whole morning on this here

https://github.com/spacehuhn/ArduinoARPspoof/blob/master/ARPspoofer/ARPspoofer.ino

LZ2DMV, 2024-09-03:
> In a funny way I spent the whole morning on this here
>
> https://github.com/spacehuhn/ArduinoARPspoof/blob/master/ARPspoofer/ARPspoofer.ino

Yes, with ENC28J60, it is definitely possible, becsuse that is a highly-configurable Ethernet controller with an SPI interface, but is it possible with an ESP32? :)

bmorcelli, 2024-09-04:
It is possible, I have it running on a project with some friends.. 

We are trying to make esp32 work as a proxy to properly perform the MITM attack, but we are stuck in this part..

We have successfully changed the ARP table of a desired target (and can be done to all targets as well)

LZ2DMV, 2024-09-04:
I think that would be very useful to have in Bruce, even as a PoC at this point. I don't think any other firmware implements ARP poisoning.

closed 2024-12-31 by bmorcelli