BruceDevices/firmware #249: CC1101 + SD card support for M5stick C Plus2

geo-tp, 2024-09-07:
**Describe the bug:**
When using a CC1101 module with an SD card on the M5Stick C Plus2, only one of the devices works correctly. Either the SD card or the CC1101 module works, but never both at the same time.

**Steps To Reproduce:**
- Connect both the SD card and the CC1101 module
- Try RF Scan/Copy, it will not work
- Remove the SD card
- Try RF Scan/Copy, now it works,but you can't save the signal on your SD card, even if you replug it

**Expected behavior:**
Both the SD card and the CC1101 module should work together on the same SPI bus

**Actual behavior:**
When one device is in use, the other stops functioning.

eadmaster, 2024-09-07:
afaik this would require an additional free gpio pin for the Chip Select , which the M5stick lacks.

As alternative, the CC1101 can also [operate without the GDO0 pin, but some features won't work this way](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/31).

GoldenboyITA, 2024-09-08:
Can we add a phisical switch between SD card reader and C1101?!

bmorcelli, 2024-09-08:
There are a few things to consider about the code, itself, that I'll post tomorrow to allow CC1101 and SdCard coexisting together..

But the problem is that once the SdCard is mounted, it is draining the dgo0 signals from CC1101 (for reading/scanning) preventing the stickC to read the signals, and it will only reset (for now) ir you take the SDCard physically out and reconnect it..

You could add a 3 pin button, with NC connected to the 3,3V and NO connected to GND and COM connected to SdCard Vcc pin, when btn is pressed, it connects the Vcc pin of the Sdcard reader to GND, resetting the circuit..

Tomorrow I will keep studying this issue to try to find out a way to  "unmount" the SD card without mods, or proposing other connection scheme

geo-tp, 2024-09-08:
> Tomorrow I will keep studying this issue to try to find out a way to "unmount" the SD card without mods, or proposing other connection scheme

That would be great, otherwise CatHack's behavior allows for recording once (and resetting to record again), and also enables sending .sub files from SD card. This allows access to all features even if a reset is needed to be able to record/save again.

bmorcelli, 2024-09-09:
Saving for later research..

Maybe it'll be needed to change cc1101 clk pin to avoid SDCard from backing to initial state

https://github.com/greiman/SdFat/issues/351#issuecomment-990970280

H3llawts, 2025-01-24:
Not sure if this could be of any help, but had my stick c plus 2 apart while working on another project and noticed free pins GND,5Vo,SDA(G21),SCL(G22), I soldered a groove port on and modified the case, but would it be possible to change the code for the cc1101 to use those 2 pins instead of G26, G25?

bmorcelli, 2025-02-17:
IT IS COMMING, BOYS!!!

CodeHedge, 2025-02-23:
> IT IS COMMING, BOYS!!!

YES!

bmorcelli, 2025-03-07:
For it to work, you need to use this schematic:
[image]

This is different from the previous connection scheme, regarding Miso, Mosi, CS and GDO0 pins..

The "legacy" connection cannot be used with SD, and will cause problems (restartings and etc)

Alberto0581, 2025-03-07:
Good afternoon, thank you for your answer, yes, it is the scheme that I have used, I am learning programming, I work in a university related to electronics. The scheme (diagram) works well, the issue is its functionality as I mentioned before, I must emphasize that I am using a somewhat old 4Gb SD, I do not know if it can influence. I can try with other SDs and partition them.

Alberto0581, 2025-03-10:
Good morning, I have changed the transistor for a BC337 and the same problem continues to occur. I have uploaded a video to Youtube so you can see what is happening, because I don't know how to explain it. Thank you very much and best regards.

https://youtube.com/shorts/L6GgbK09LRM?si=IYiWgNfFdrXfRLmc

Alberto0581, 2025-03-27:
Good morning, I continued testing today. The way I've gotten it to work without having to remove the SD card is by placing a microswitch from an old mouse between the positive terminal of the card reader, between the common terminal and the normally closed terminal. When I need to operate the card, such as /config/rf module/ CC1101 shared, custom subGhz, or saved, I press the microswitch and the operation is completed correctly.
I'm uploading a demonstration video. Thanks and best regards.

https://www.youtube.com/shorts/wvPrE5VQAm0
[https://ibb.co/BHWN7p9T](url)

closed 2025-03-28 by bmorcelli

bmorcelli, 2025-05-07:
The SD Card breakout doesn't have a Pullup resistor on the CS Line, need to add a 10k Resistor connected between CS and 3,3V

[image]

mohzart, 2025-05-22:
Hi there :) I'm a noob... sr ;) My questions are

1. Howcome the "CatHack" firmware works alright (correct me if i'm wrong) with "simple-ish" connections between/legacy connections using C1101 in sub-Ghz stuff while when using "Bruce" the connection has to be made via transistors/resistors and difficult (for me) to put together package.

2. Does this setup work with NRF24 too as it in "CatHack"? What's the diagram (if there is any) for NRF24+SD, or what is the diagram for it w/o SD?

3. Will this diagram made by Pirata be a permanent solution or will there eventually be an easier way to connect M5+SD+CC1101/NRF24

 
Thanks in advance,
(sorry for noob-ish questions)
R.

CodeHedge, 2025-05-22:
[image]

mohzart, 2025-05-22:
I don't fully understand... 

does Cathack's solution requires only restarting the stick after putting the SD inside, and then it works (i.e. you can transmit recorded files from SD via CC1101) - because that 5-6sec is a low price to pay + how often the SD card is being replaced?

OR

after mounting SD it crashes/restarts(?) (and the CC1101/whole package) simply does not work properly? In such a case I don't know why Cathack would release that diagram as a working solution

OLSO
(just asking for predictions, i know that it's a WIP)
+ will that solution with the transistor be a permanent one?
+ will it work with NRF24 + SD?

thanks in advance :)

p.s. Below photos of my "spaghetti-ish solution" in progress - i'm not the best as you can see ;P

[image]
[image]

bmorcelli, 2025-05-22:
> does Cathack's solution requires only restarting the stick after putting the SD inside, and then it works (i.e. you can transmit recorded files from SD via CC1101) - because that 5-6sec is a low price to pay + how often the SD card is being replaced?

It is a CatHack workaround that doesn't work for bruce, because Bruce mounts the SDCard at the beginning to provide lots of other features, and we won't change the firmware just for that... You can use bruce with this setup, but you will need to be constantly plugging and unplugging the sdcard from the slot, and vulnerable to many errors and problems

And plus, it is not a reliable solution, electronically speaking...

> after mounting SD it crashes/restarts(?) (and the CC1101/whole package) simply does not work properly? In such a case I don't know why Cathack would release that diagram as a working solution

The solution works, but you need to be turning on and off your device everytime you save/read a file on SDCard and try to send it with the CC1101.

> + will that solution with the transistor be a permanent one?

Yes, it is bruce solution, and it is permanent for this device.. electronically stable and provides everything bruce needs with the cost of 2 resistors and one transistor (less than 2 dollars in a electronic store)

> + will it work with NRF24 + SD?

Yes, it will.

mohzart, 2025-06-04:
Solved :) ...I made it from scratch (WITH THE SMALLER SD MODULE) and now it is fully operational. I guess that bigger one was "blocking/sucking" power or smth.

The only (ultra minor issue I encountered is that when switching from NRF to CC [or vice versa] I have to reboot)

Thanks for the help, diagrams etc
R.

P.S. Now to do some evil... muahaha (OFC: joking ;))

mohzart, 2025-06-04:
[image]

[image]

kriku, 2025-06-09:
> I dont know if it will help but try to GND the SD and nrf24 on same pin...
> If u dont do that already ....
> […](#)

Actually this is setup with cc1101 module.

kriku, 2025-06-09:
I think I found it.

[image]

There is [header 4](https://m5stack-doc.oss-cn-shenzhen.aliyuncs.com/512/Sch_M5StickC_Plus2.pdf) I guess.

IO22 and IO21 respectively. Near the IR transceiver.

I think it's possible to cut m5stick case and solder pinout for this header. If it's not used by anything.

Spoiler - it used by internal imu. 

Do anybody know if there are unused mcu pins, to which we can connect?

Huzzla101, 2025-06-09:
What you could do is to get a SD SNIFFER , the red ones and enter into SD
card then you will get a new SD card AND 10new pinouts to use...... And the
sd card will store from booth the devices the SD sniffer is put into and
the device connected to the 10 pinouts will also be stored....

 But you say  " Actually this is setup with cc1101 module."  .. CC1101 and
nrf24 have the same set up for pinouts on  mostly all devices....And i have
made it work before using the same SD card sniffer as you , it has 6
pinouts correct..... and the nrf has 8 where u need use 7 of them  and it
has the possibility to let the ground be used 2 times just attach it to the
little metal part you have at the input of the nrf24 GND wire and u may not
need to re attach ur sd card each time... I would also suggest using the
last pinout on the nrf24 that is not used , to use that also as a second
ground for the nrf24 sniffer then attach the first ground pin besides the
voltage to the sd card so it get the first real GND setup mixed in with
nrf24 ... But hey  just a suggestion  ... I hope for the best =)