BruceDevices/firmware #251: Sorce code reported as a virus by Cisco Secure Endpoint.

mtclpt, 2024-09-08:
Hi,
Cisco Secure Endpoint report as a malicious file (E:\bad_usb.cpp).
The cause is the payload example at the beginning of the file.
My suggestion is to remove that example from the file.
Message: Quarantine of malicious file (E:\bad_usb.cpp) successful.

Ademolght, 2024-09-08:
@mtclpt Just delete cisco secure

IncursioHack, 2024-10-24:
> Olá, o Cisco Secure Endpoint relata como um arquivo malicioso (E:\bad_usb.cpp). A causa é o exemplo de payload no início do arquivo. Minha sugestão é remover esse exemplo do arquivo. Mensagem: Quarentena de arquivo malicioso (E:\bad_usb.cpp) bem-sucedida.

When using resources that involve hacking, if the machine has antivirus, EDR and the like, they will probably hinder you. Unfortunately, we cannot guarantee that part of Bruce's code will not be interpreted maliciously by the tools I mentioned above. My suggestion is that if you are downloading and modifying code, that you have an appropriate environment for this, a virtual machine or even Docker would solve your case.

closed 2024-10-24 by IncursioHack