BruceDevices/firmware #259: frequency detection

cinouw, 2024-09-12:
Hi bruce

the other day i was trying a io433 firmware on m5 stick c plus 2 and it have a really cool feature. The feature is basically an option that detects the frequency that the subGhz remote/device uses i think it will be very usefull even for diffrenf 433.xx devices for detecting the frequency and configuring subGhz rx/tx settings accordingly.

Thanks for the greate work 

Help offer : I have lots of development modules in hand if you need any help for testing I can do it

eadmaster, 2024-09-13:
should be doable reusing [code from here](https://github.com/mcore1976/cc1101-tool/blob/main/cc1101-tool-esp32.ino#L480) , please post some shots on how the UI should look like, i think i'll add the backend logic only for this.

eadmaster, 2024-10-04:
added in [v1.6](https://github.com/pr3y/Bruce/releases/tag/1.6), this can be closed

closed 2024-10-05 by bmorcelli