BruceDevices/firmware #260: help offer :)

cinouw, 2024-09-13:
Hi Bruce team I really liked the project and I would like to help you with it so I have lots of development modules in hand and i can help you with testing if needed

thanks for the greate work

module list
cardputer x2
stick c plus 2
cc1101 x3
nrf24l01 x2
sd sniffer
arduino pro micro x3
arduino uno
sd card module
neo 6m x2
esp32x3
esp8266
pn532
mcrf522
rdm6300
rfid 2 module
ir rx/tx module
433.62mhz rf rx/tx module
I2C expansionhub
I/O expansionhub
gps/bds module

bmorcelli, 2024-09-16:
Hey @cinouw, we appreciate your offer!!

We need some development on Serial UART communication.. all these devices might help at some point, at least in a testing phase..

Thank you for following the project and provide insights about its development.. If you are not participating on Our Discord server, there is a good place to discuss things, help people and learn some new things about esp32 and m5 devices

cinouw, 2024-09-16:
Thanks for the replay

I didn't know about the discord server i will be also participating there

closed 2024-10-24 by IncursioHack