BruceDevices/firmware #263: Run JS from context menu

rennancockles, 2024-09-13:
**Describe the bug:**
Running JS interpreter from the sdcard context menu makes Bruce restart. Running it from the Interpreter menu works fine though

**Steps To Reproduce:**
Others -> SD Card -> Anything.js -> JS Script Run
- Device used: Cardputer

**Expected behavior:**
Run JS interpreter with selected file

**Actual behavior:**
Restarts the system

**Additional context / logs:**
```
Selecionado JS Script Run
Read from file
loaded file:
var screenWidth = width();
var screenHeight = height();
var battery = getBattery();
var board=getBoard();
var redraw=true;

setTextSize(2);
function updateBtn() {
    if(getSelPress() && getNextPress()) {
        drawFillRect(0, 0, screenWidth, screenHeight, color(255, 0, 0));
        setTextColor(color(0,0,0));
        drawString("Exiting", screenWidth / 2 - 60, screenHeight / 2 - 10);
        delay(3000);
        exit();
        throw new Error("Exit");
    }
    if(getPrevPress()) { 
        setTextColor(color(255,0,0));
        drawFillRect(0, 0, screenWidth, screenHeight, color(0, 0, 0));
        drawString("Prev Btn Pressed ", 10, screenHeight/2-8);
        redraw=true;
    }
    if(getSelPress()) { 
        setTextColor(color(0,255,0));
        drawFillRect(0, 0, screenWidth, screenHeight, color(0, 0, 0));
        drawString("Selec Btn Pressed", 10, screenHeight/2-8);
        redraw=true;
    }
    if(getNextPress()) { 
        setTextColor(color(0,0,255));
        drawFillRect(0, 0, screenWidth, screenHeight, color(0, 0, 0));
        drawString("Next Btn Pressed ", 10, screenHeight/2-8);
        redraw=true;
    }
    if(redraw==true) {
        battery = getBattery();
        drawString("Battery level: " + battery, 10, 10);
        drawString("Board: " + board, 10, 27);
        drawString("Sel + Next to Exit", 10, screenHeight-22);
        drawRect(5,5,screenWidth-10, screenHeight-10, color(150,20,210));
        delay(200);
        redraw=false;
    }

}

while (true) {
    updateBtn();
}
Guru Meditation Error: Core  1 panic'ed (Double exception). 

Core  1 register dump:
PC      : 0x40381f9a  PS      : 0x00040436  A0      : 0x820c0480  A1      : 0x3fcea030  
A2      : 0x00040436  A3      : 0x00040026  A4      : 0x00000001  A5      : 0xbea6ae50  
A6      : 0x397f5250  A7      : 0x3fcd9b24  A8      : 0x3fcea0f0  A9      : 0x00000000  
A10     : 0x3fcd9b24  A11     : 0x0000001e  A12     : 0x3fceb170  A13     : 0x00000000  
A14     : 0x7ff00000  A15     : 0x3fcd9b24  SAR     : 0x0000001d  EXCCAUSE: 0x00000002  
EXCVADDR: 0x00000000  LBEG    : 0x420d104c  LEND    : 0x420d105e  LCOUNT  : 0x00000000  

Backtrace: 0x40381f97:0x3fcea030 0x420c047d:0x3fcea0f0 0x40381f97:0x3fcea110 0x4037407d:0x3fcea020 0x40378ef9:0x3fcea050 0x40378f22:0x3fcea070 0x40388675:0x3fcea090 0x420b11e1:0x3fcea0b0 0x420bf029:0x3fcea0d0 0x420bfb6e:0x3fcea0f0

ELF file SHA256: d427a3f5b02b4c72

Rebooting...
```

eadmaster, 2024-09-15:
it is currently working with my fork:
https://github.com/eadmaster/Bruce/tree/interpreter-additions

I'll send a pr later for this with other fixes:

bmorcelli, 2024-09-16:
The stack usage of interpreter() function is absurd, and for this is recommended to call this function from loop()

So I would suggest to prepare the global vars and let it run in the main:

bjs_interpreter:
```
bool run_bjs_script_headless(String code) { //can be deleted
    script = code;
    returnToMenu=true;
    interpreter_start=true;
    //while(interpreter_start) {
        interpreter();
    //    delay(1);
    //}
    interpreter_start=false;
    return true;
}

bool run_bjs_script_headless(FS fs, String filename) {
    String code = readScriptFile(fs, filename);
    returnToMenu=true; // return to while in loop() in main.cpp
    interpreter_start=true; // trigger the jump out from while loop to star interpreter here https://github.com/pr3y/Bruce/blob/7a8850c29f8c5b7790f446578335580aa5675522/src/main.cpp#L311
    return code>0 ? true:false;
}
```

This way the code will be called from loop() here https://github.com/pr3y/Bruce/blob/7a8850c29f8c5b7790f446578335580aa5675522/src/main.cpp#L296

It would be better to call it in a new task and "freeze" looptask at this point, but I didn't have time for this yet.

eadmaster, 2024-10-04:
@rennancockles: pls retest, it should be fixed in  [v1.6](https://github.com/pr3y/Bruce/releases/tag/1.6)

closed 2024-10-05 by bmorcelli