BruceDevices/firmware #265: Pwnagotchi

disaxq, 2024-09-14:
**Describe the bug:**

when my m5stick plus 2 detects pwnagothi, then it just reboots

Ademolght, 2024-09-14:
me too on cardputer and interpreter also reboot

bmorcelli, 2024-09-16:
> **Describe the bug:**
> 
> when my m5stick plus 2 detects pwnagothi, then it just reboots

Can you please send the Serial Monitor log for us to understand the problem and address the point to fix?

disaxq, 2024-09-16:
how i can it?

disaxq, 2024-09-16:
> Соавтор

how i can it?

bmorcelli, 2024-09-16:
Plug your stickCPlus2 into a computer, with the cable you used to flash, open m5burner and click at the 3 bars at the upper right corner, select serial Monitor, select the COMx that your device is connected (or plug into USB at this moment), hit connect and log the things...

Make the error happen and send the log

disaxq, 2024-09-16:
> Подключите stickCPlus2 к компьютеру с помощью кабеля, который вы использовали для прошивки, откройте m5burner и нажмите на 3 полоски в правом верхнем углу, выберите serial Monitor, выберите COMx, к которому подключено ваше устройство (или подключите к USB в данный момент), нажмите connect и внесите изменения в журнал...
> 
> Исправьте ошибку и отправьте журнал

Selecionado Pwnagotchi
Pwnagotchi Initialized

Stack smashing protect failure!

Backtrace: 0x40084691:0x3ffdde70 0x40093cad:0x3ffdde90 0x40083a4a:0x3ffddeb0 0x4010768d:0x3ffdded0 0x401077ae:0x3ffddf40 0x40104992:0x3ffddfb0 0x40104afa:0x3ffddfd0 0x400d8f83:0x3ffde0f0 0x400d4fed:0x3ffde110 0x400d96c6:0x3ffde180 0x40309645:0x3ffde260 0x400f4d73:0x3ffde280 0x40197510:0x3ffde2d0

ELF file SHA256: d3535330a8103cc6

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E ������J*��}�х}���� not found otadata
[    37][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    64][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    73][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    84][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[    99][D][main.cpp:316] setup(): Total heap: 194260
[   104][D][main.cpp:317] setup(): Free heap: 159292
[   109][D][main.cpp:320] setup(): PSRAM Not Found
[   113][D][main.cpp:321] setup(): Total PSRAM: 0
[   118][D][main.cpp:322] setup(): Free PSRAM: 0
[  1359][I][main.cpp:237] load_eeprom():   
*-*EEPROM Settings*-*   
- rotation  =003,   
- dimmerSet =010,   
- Brightness=100,   
- IR Tx Pin =019,   
- IR Rx Pin =033,   
- RF Tx Pin =032,   
- RF Rx Pin =033,   
- Time Zone =003,   
- FGColor   =0xA80F   
- RfModule  =000,   
- RfidModule=000,   
*-*-*-*-*-*-*-*-*-*-*
[  1389][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 255
[  1398][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000

disaxq, 2024-09-19:
Selecionado Pwnagotchi
Pwnagotchi Initialized

Stack smashing protect failure!

Backtrace: 0x40084691:0x3ffdde70 0x40093cad:0x3ffdde90 0x40083a4a:0x3ffddeb0 0x4010768d:0x3ffdded0 0x401077ae:0x3ffddf40 0x40104992:0x3ffddfb0 0x40104afa:0x3ffddfd0 0x400d8f83:0x3ffde0f0 0x400d4fed:0x3ffde110 0x400d96c6:0x3ffde180 0x40309645:0x3ffde260 0x400f4d73:0x3ffde280 0x40197510:0x3ffde2d0

ELF file SHA256: d3535330a8103cc6

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E ������J�*��}�х}���� not found otadata
[ 37][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[ 64][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[ 73][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[ 84][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[ 99][D][main.cpp:316] setup(): Total heap: 194260
[ 104][D][main.cpp:317] setup(): Free heap: 159292
[ 109][D][main.cpp:320] setup(): PSRAM Not Found
[ 113][D][main.cpp:321] setup(): Total PSRAM: 0
[ 118][D][main.cpp:322] setup(): Free PSRAM: 0
[ 1359][I][main.cpp:237] load_eeprom():
-EEPROM Settings-

rotation =003,
dimmerSet =010,
Brightness=100,
IR Tx Pin =019,
IR Rx Pin =033,
RF Tx Pin =032,
RF Rx Pin =033,
Time Zone =003,
FGColor =0xA80F
RfModule =000,
RfidModule=000,
----------*
[ 1389][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 255
[ 1398][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000

bmorcelli, 2024-09-20:
Can you test with the artifacts of this Pull request?

https://github.com/pr3y/Bruce/actions/runs/10961481295

closed 2024-10-05 by bmorcelli