BruceDevices/firmware #270: Fn button LED for cardputer

edeot, 2024-09-17:
Hello Ive had some issues while using Bruce Firmware, sometimes when i was trying to select a folder or document on the "Cardputer" hardware- it would offer to make a new folder or rename, which i originally thought was a bug. 
After a day it occurred to me maybe the fn Button was activated while selecting the folder, and that was indeed my issue i was having- i needed to deactivate the fn button in order to select the document.

the issue was there was no indication that the "fn" button on the keyboard was activated

**Describe the solution you'd like**
I thought it would be great if one of the LEDs on the device could be actived when the fn button is active. (ideally orange light to go with the color on the keypad)

**Additional context**
I understand this is only a "cardputer" device only issue, so it may not be of importance

**also, can we put the BADUSB feature on the main menu?**

bmorcelli, 2024-09-17:
The fn button doesn't do anything in the project, the behavior you are describing is related to short and long press detections on file selections..

It counts 200ms to identify if open menu contest (btn is still pressed after 200ms) or open the folder (not pressed after 200ms)..

eadmaster, 2024-09-19:
> can we put the BADUSB feature on the main menu?

it's in the ["Others" menu](https://github.com/pr3y/Bruce/wiki/Others#badusb).

closed 2024-09-20 by bmorcelli