BruceDevices/firmware #283: Custom Boot Image on m5stick c plus 1.1 doesn't work

logm1lo, 2024-09-26:
**Describe the bug:**
Even if I rename the file to boot.jpg and push it to the root folder in LittleFS, it still doesn't work

**Steps To Reproduce:**

- Device used: M5STICK C PLUS 1.1 
- Compiled and flashed the latest firmware using platformio and esptool

**Expected behavior:**

- Custom image on boot

**Actual behavior:**

- There is nothing on the screen with only the Bruce + version on boot

bmorcelli, 2024-09-26:
Send me your file, if you can, for me to test.

logm1lo, 2024-09-28:
> Send me your file, if you can, for me to test.

do you mean the folder or the bin file?

logm1lo, 2024-09-28:
sorry for the late reply

logm1lo, 2024-09-28:
[https://1drv.ms/f/s!An_AR-itx59FgrREkvVlZ-jFqSzbeA?e=MJxb4k](url)

this is a onedrive link to a folder that has my Bruce folder and the binary that I've compiled in there.

bmorcelli, 2024-09-30:
The jpg file you tried..

the project I have here...

Try installing the project from https://bruce.computer/ Web Flasher (Beta Version)

and upload one of the files from https://github.com/7h30th3r0n3/Evil-M5Core2/tree/main/SD-Card-File/img 
and rename the file to "boot.jpg"
and save into the root of your SD Card.

IncursioHack, 2024-10-24:
The last comment with the solution was presented 3 weeks ago and we had no response, so I will be closing this issue.

closed 2024-10-24 by IncursioHack