BruceDevices/firmware #285: RF - Frequency Analyzer

sousaaguilherme, 2024-09-27:
Hi,

I've just setup a CC1101 with bruce and it seems to work.

Is there a plan to add a frequency analyzer so that we can determine the frequency we should configure to clone something if we don't know the frequency?

Thank you for your work!

bmorcelli, 2024-09-27:
Yes, it'll come next week

sousaaguilherme, 2024-09-27:
Thank you so much! I'll just close this

closed 2024-09-27 by sousaaguilherme

eadmaster, 2024-09-28:
related: https://github.com/pr3y/Bruce/issues/259

ok123de, 2025-05-20:
its been 7 months

> Yes, it'll come next week

Huzzla101, 2025-05-20:
AHHH   an Frequency Analyzer ... That would be awesome.... A
analyzer/sniffer that will sniff ALL SubGhz signals beeing sendt in all 3
ranges in real time continuously...That is something we must have.. Who
have the needed skills for that? VADIM/VAD7......Have you comed around
yet... ? =)

Also we need a hopper -its a simple add on to RF reader that makes you
choose if you want only to scan one range or one specific Ghz that we
already have, BUT a hopper is simply the possibility to turn it on so it
will only scan your already stored RF signals, its VERY USEFULL for the Pen
Testing areas to get a clear overview of your surroundings  and it
evolvements...