BruceDevices/firmware #288: CC1101 and NRF24L01 in one

M1cr0K1ller, 2024-09-28:
My question is it possible to attach here a NRF24L01 
[image]
With 2 switches on MOSI and MISO that will eighter go to CC1101 or NRF24L01.
If it's possible, could some one do the schematic, cause i don't know how to do it right

bisaw37, 2024-09-28:
Yes you totally can, and I’m planning to do something similar, but much simpler. Instead of 2 switches, just plug everything in together, and put one switch between the VCC of both of them. Only 1 switch to fumble around with, and it’s also scaleable. I’m going to do this with the cc1101, nrf24l01, pn533, and m5stack IR module.

M1cr0K1ller, 2024-09-28:
That's actually a lot smarter. Thanks a lot!

closed 2024-09-30 by bmorcelli