BruceDevices/firmware #290: WiFi Spy Microphone

FernandoJose2, 2024-09-29:
What are your comments about this feature? I think, it will be very cool in James Bond 007 movies.

We no need a external module for this. Cardputer already has a esp32, sdcard module and mic.

WiFi Spy Microphone With ESP32

https://www.instructables.com/Spy-Microphone-Connected/

https://www.hackster.io/mvtdesign/wifi-spy-microphone-with-esp32-devkit-a02120

https://www.hackster.io/mvtdesign/esp-wifi-spy-microphone-esp32-and-esp8266-80d3a5

https://wicard.net/esp-spy-microphone/

LEGOATDU49, 2024-09-29:
Yes, like this we could hide the device in a place and hear the person live via the webui or record the audio on the SD card

uncooldude, 2024-12-29:
Agreed, We really need this feature, u can make different versions of bruce, like burce-pentesting core (which will have the pentesting tools), and other distro with smthing crazy like this