BruceDevices/firmware #294: Feature request: repeated signal sending

bisaw37, 2024-09-30:
I think it would be very nice to have a way to, when sending anything via RF or IR, repeatedly send a signal over and over until manually stopped. You could place this in the menu when you choose a file to send via SD card or littleFS, or in the actual send area for RF or IR (both would be better for redundancy). This isn't really a very necessary thing to have but it could be useful in many circumstances.

eadmaster, 2024-10-06:
this can easily be done via a [js script](https://github.com/pr3y/Bruce/wiki/Interpreter)

e.g.

````js

curr_filename = dialogPickFile("/");
repeats = keyboard("1", 99, "repeats?");
repeats_int = Number(repeats);  // parseInt() also works 
if(repeats_int != NaN) {
for( var i=0 ; i < repeats_int; i++ ) {
    irTransmitFile(curr_filename);  // change to subghzTransmitFile for RF
 }
}

````

IncursioHack, 2024-10-24:
In view of the solution presented by @eadmaster, I will be closing this issue.

closed 2024-10-24 by IncursioHack