BruceDevices/firmware #305: Flipper like JS format

MegaTheLEGEND, 2024-10-04:
**Describe the solution you'd like**

It would be Cool if the JS interpreter followed the same format as that of the Flipper.

**Additional context**

instead of this
```

wifiConnect(ssid : string);

dialogError(msg : string);
```

it would be like this
```
let math = require("math");  // math modual would be cool. (maybe it already exists?)
let badusb = require("badusb");
let wifi = require("wifi");
let dialog = require("dialog");

math.abs(-5); // 5
badusb.setup();

if (badusb.isConnected()) {
    badusb.press("E");
} else {
   dialog.error("failed to connect badusb"); //  Show an error
}

wifi.connect(ssid : string);
```

This provides a more organized method of calling functions and also would increase possible interoperability between Bruce and Flipper scripts

pr3y, 2024-10-04:
This is already implemented in the firmware.
Read here https://github.com/pr3y/Bruce/wiki/Interpreter

eadmaster, 2024-10-04:
i've investigated this a bit already, it needs [this optional extra](https://github.com/joeqread/arduino-duktape/tree/main/extras/module-node).

MegaTheLEGEND, 2024-10-04:
That would be cool. Also I feel like it would be great if users could share their js apps on a catalog on the website. A lot of times the most innovative and creative ideas come from the users and community.

pr3y, 2024-10-05:
Loved this idea! this will probably be added soon!

closed 2024-12-07 by bmorcelli