BruceDevices/firmware #307: Bug: boot screen shows "3813139" instead of 1.6

bisaw37, 2024-10-04:
**Describe the bug:**
For whatever reason, whenever I boot up Bruce on my stickCplus2, it displays "3813139" instead of 1.6 where the version is.

**Steps To Reproduce:**
1. install Bruce 1.6.
2. boot it up and look at the boot screen.
- Device used: m5stickC plus2

**Expected behavior:**
It would show "1.6"

**Actual behavior:**
It displays "3813139"

**Additional context / logs:**
This certainly isn't at the top of the bug fixing list, but I'm just placing it here so its known. 
And also I run m5launcher and installed Bruce there, if that's important.

Ari-apk, 2024-10-04:
its shows that when its installed via m5burner so if you install it from [https://bruce.computer/flasher](url) it will show 1.6

bisaw37, 2024-10-04:
But if I install there wont it overwrite m5launcher with bruce?

DrShrike, 2024-10-04:
> But if I install there wont it overwrite m5launcher with bruce?

Yes, it would install Bruce as the only firmware.
Doesn't seem to be a huge issue though, you probably don't need to reinstall.

bisaw37, 2024-10-04:
Ill just stick with what I have, maybe eventually it'l get fixed. After all it isn't too much of a problem to forfeight m5launcher over.

pr3y, 2024-10-05:
By mistake i uploaded the Beta bins to M5burner, this number is the initial digits for commit hash.
Now it should be right on burner also, just flash again your devices.